daniel5151/gdbstub #3: Publish to crates.io

sapir, 2020-03-26:


daniel5151, 2020-03-26:
Thanks for the interest!

While there's still quite a bit of cleanup / refactoring I'd like to do to `gdbstub` (including a large refactor that removes the dependency on `alloc` entirely), I certainty don't mind publishing "gdbstub 0.1" to crates.io.

The only real blocker to publishing the crate is missing docs and examples.  Thanks to COVID-19, I've suddenly got lots of free time on my hands, so I should be able to do a documentation pass in the next couple weeks.

sapir, 2020-03-26:
Thanks for making this!

fwiw there are "examples" at:
* https://github.com/michelhe/rustboyadvance-ng
* https://github.com/sapir/microcorruption-emu

daniel5151, 2020-03-29:
https://crates.io/crates/gdbstub

closed 2020-03-29 by daniel5151