daniel5151/gdbstub #10: Trait objects in public API

Plecra, 2020-07-12:
Would it be possible to use generic parameters for the public trait methods?

```rust
fn resume(
    &mut self,
    actions: impl Iterator<Item = (TidSelector, ResumeAction)>,
    check_gdb_interrupt: impl FnMut() -> bool
) -> Result<(Tid, StopReason<<Self::Arch as Arch>::Usize>), Self::Error>
```

Personally, I find this easier to read, but it also allows consumers to use a monomorphised version of the function. I think this is useful for more constrained debugger implementations since it allows the generated code to be much smaller for calls like `resume(Empty, || false)`.

I'm not sure why you'd need a `Target` trait object, but there's no reason you couldn't make one internally: https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=4e94a730ae674c63595bed600b64cec7

daniel5151, 2020-07-12:
That's a clever trick for making an object-safe trait from one with generic parameters!

Funny enough, I actually switched from generic parameters to dynamic dispatch fairly recently, in 55fd99c. As for why you'd want a `Target` trait object: consider a multi-system emulator where different systems share a core architecture, but differ in connected hardware. If that seems like a oddly-specific example, that's because I'm planning on doing something like that in my other project, [`clicky`](https://github.com/daniel5151/clicky).

There is a small issue with this approach though: the actual `actions` iterator changes depending on what sort of packet GDB sent. See https://github.com/daniel5151/gdbstub/blob/fab622d/src/gdbstub/mod.rs#L394-L443. I'm fairly sure that switching to `impl Iterator` in this case would actually lead to _more_ code than the dynamically dispatched alternative (generating a monomorphised version of the potentially heavy `resume` method for each variation).

That said, I'm fairly sure that most of the `&mut dyn FnMut`s wouldn't have the same issue.

---

To be entirely honest, I haven't actually had the chance to test `gdbstub` in a proper `#![no_std]` setting just yet, and as such, I haven't put too much work into optimizing the library for code size. I'm sure there is _plenty_ of fat to trim from `gdbstub`, including these dynamic dispatch calls.

Plecra, 2020-07-13:
I see! I had the impression that users might be calling this trait themselves, but it looks like it's really just for internal use. Personally, I think taking an `Actions` struct is the best option here, since it gives you control over the implementation details (It can just contain that `&mut dyn Iterator`, or be implemented with an enum if that helps performance), and makes it clear that your library is meant to be the source. It makes the problem of implementing the trait much simpler for users if they know they don't have the responsibility of constructing the type somewhere too.

daniel5151, 2020-10-10:
Quick update.

While working on version `0.4.0`, I spent quite a bit of time optimizing `gdbstub`'s binary size. As far as I can tell, using trait objects is fine, since most of the time the trait object gets entirely devirtualized and/or inlined, making it equivalent to the generic implementation with the added benefit of being implicitly object safe.

That said, I did notice that the compiler had a hard time inlining / optimizing the `GdbStub::do_vcont` method, which results in bunch of somewhat heavy closure+iterator code in the resulting binary. I haven't looked into it _too_ much, but I have a feeling it's partially due to the fact that `GdbStub::do_vcont` is called from the `vCont`, `c`, and `s` packet handlers, meaning that it never gets inlined. Unfortunately, simply slapping a `#[inline(always)]` doesn't help, so the issue is probably a bit more subtle than that...

I've gone ahead and removed the `help wanted` label, as this is something I'll probably look into at some point myself (though that's not to say I wouldn't appreciate a PR addressing it!)

I'll still leave this issue open, as it'd still be nice to explore any alternatives and get some hard data.

daniel5151, 2021-11-16:
`gdbstub` has been using the IDET approach in production for quite a while now, and myself and others have confirmed that the current use of dynamic dispatch in `gdbstub` doesn't incur any serious overhead in most use cases, and in fact, often results in _tighter_ code thanks to the aggressive inlining / devirtualization / dead code elimination when compiled in release mode.

As such, I'm (finally) going to close this issue :)

closed 2021-11-16 by daniel5151