daniel5151/gdbstub #26: Explore + Document how to work with Big-Endian (and Bi-Endian) Targets

daniel5151, 2020-09-04:
https://github.com/daniel5151/gdbstub/pull/25 (which added support for 32-bit PowerPC architectures) raised some interesting questions regarding how `gdbstub` should handle big-endian and bi-endian targets.

Namely, should the `Registers::gdb_de/serialize` methods be configurable based on the target's current endianness? i.e: plumb through some sort of `Target` API which queries the endianess of the target, and switches between `to/from_be_bytes` and `to/from_le_bytes` appropriated?

Unfortunately, this isn't something I can easily test, since I don't have any bi-endian targets to play with...
As such, I'm marking this issue as "help wanted," so that hopefully someone who _does_ have experience with bi-endian systems could clarify the situation.

jamcleod, 2020-09-04:
When you say "bi-endian" are you referring to runtime-configurable byteorder (as I would interpret it) or fixed-endian systems which can be either ordering? As I feel like each of those would warrant different solutions.

daniel5151, 2020-09-05:
From the [wikipedia article on Endianness](https://en.wikipedia.org/wiki/Endianness#Bi-endianness):

> Note that the term "bi-endian" refers primarily to how a processor treats data accesses. Instruction accesses (fetches of instruction words) on a given processor _may_ still assume a fixed endianness, even if data accesses are fully bi-endian, though this is not always the case, such as on Intel's IA-64-based Itanium CPU, which allows both.

This seems to imply that "bi-endian" is an umbrella term which encapsulates both the cases you mentioned?

Just to reiterate though, as someone who don't have _any_ hands-on experience working with bi-endian and big-endian systems, I'm not at all sure what the best approach would be in this case (hence the "help wanted" tag on the issue).

xobs, 2025-05-13:
I believe that with bi-endian systems, GDB will look at the endianness of the function, or possibly the endianness of any status register it has that it can learn from, to figure out what the endianness is. For example, if I wiggle bit 9 on the CSPR on an ARM chip then the endianness setting in GDB changes.

This may be something that can go on the implementation, rather than in the library. Regardless, now that I'm [working on TMS570 support](https://github.com/probe-rs/probe-rs/pull/3312), which is a fixed-BE device, I'll be looking into this closer. Though it may very well be that I just end up wrapping things that get delivered to gdbstub, and treat it as an LE system.

An interesting note: while the TMS570 is BE, the ADIv5 debug interface is LE. So it's a bit of a hybrid. So things need to get swapped all over the place.

daniel5151, 2025-05-22:
Thanks for the insights and experimentation here @xobs!

One thing I'd like to understand is how the GDB client determines the endianess of the target it has connected to (i.e: what results in `show endian` reporting that the target endianess is set automatically?). 

Is this information encoded in the target XML somewhere? Or is there some other metadata packet being exchanged that signals the endianess? Or does GDB know "a-priori" that certain target XML files correspond to little/big endian? Or does GDB "sniff out" the endianess based on a some heuristic determined by values returned from various memory / register queries?

> That's because I can't figure out how to update the `.endianness` field.

I'm not super familiar with `probe-rs`'s code, but it feels like this isn't the right place to stuff the endianess field. I imagine that `probe-rs` has their own "dynamic `Arch`" abstraction on-top of `gdbstub`'s compile-time "fixed `Arch`" trait, which handles all the details of reading/writing an opaque `Vec<u8>` register file that the `gdbstub` fixed `Arch` trait simply communicates to/from the client. If so - that sort of endianess byte swizzling should probably be happening at that higher level of abstraction, which should (presumably) have a more reasonable plumbing to update the endianess at runtime. I'll also note that doing `chunks_exact(4)` as you've described above is unlikely to work correctly, as it seems hard-coded to work with architectures that consist entirely of 32-bit registers.

That said, setting aside the details of how to accomplish what you're trying to do today in `probe-rs`, your example brings up a fair question of whether `gdbstub` itself should be enlightened to endianess, and have a mechanism to inform the stateful `impl Registers` block of what the current endianess is. e.g: adding a new `fn set_endian(&mut self, endian: Endianess)` function that `gdbstub` can invoke, which is wired to some user-controlled endianess notification mechanism\*.

\*which brings us back to my initial question, regarding how the GDB client coordinates with the GDB server to ascertain endianess. It also raises another question about whether it would be possible to dynamically toggle endianess at runtime, and how that might work.

xobs, 2025-05-23:
At least on ARM, GDB will set its endianness based on the value of bit 9 in `$cpsr`. Registers are never swapped, at least by the time they exit the chip, so bit 9 is always at the same location regardless of the endianness. Even within probe-rs, I need to swap memory reads but never register reads.

If I change the endianness in gdb with `set endian little`, then all the register values show up correctly but memory values become swapped. If I leave it as normal, then memory values are correct.

If I leave it as `auto` and manually swap register values (and only register values) like I do in that hack, then it works.

So at least on ARM, it consults bit 9.

Maybe we could do something fancy inside the register cache to look at that bit and convert to/from the native endianness of the host...?

xobs, 2025-05-23:
So part of the problem was an issue on my part -- probe-rs didn't delegate the `.endianness()` function so it always reported little endian.

Now that I've done that, everything is working and I'm able to use 0.7.5 without issue. I've added a function to swap register values as they're read out from the core and written back in.

Given this, I don't think there is any additional API needed from gdbstub.

daniel5151, 2025-05-23:
Thank you for sharing all these observations!

I'm glad to hear you've gotten things working in `probe-rs` without changes to `gdbstub`, but given that most other users of `gdbstub` _aren't_ relying on the probe-rs specific "dynamic arch / target" abstraction, it seems _plausible_ that the `gdbstub` may yet want to add some kind of endianess plumbing. e.g: something like:

- adding a `fn endianess(&self) -> Endianess` API to the `impl Registers` trait, which can be used to report the current endianess (either hard-coded, or by sniffing a bit in the register file)
- plumbing through the return value of `endianess` to internal APIs such as `gdb_{de}serialize`, and the various Target memory read/write APIs, allowing `gdbstub` to transparently shuffle bytes around as-needed

This is a rough sketch, as I haven't yet totally wrapped my head around the nuances around when bytes are/aren't flipped (i.e: within the stub, or when sent to GDB, or in the GDB client, etc...), but maybe these kinds of APIs could help make it easier for `Target` authors to avoid getting bogged down in manually accounting for endianess in the various register/memory read-write paths.

And really - that's the fundamental underlying question that we [folks working on `gdbstub`] need to be asking ourselves: _should_ `Target` authors who opt to use pre-existing `Arch` implementations (e.g: from `gdbstub_arch`) have an "easy" experience of being totally ignorant to the details of endianess in their system, relying on extra logic in `gdbstub` internals + the `Arch` implementation to handle the various bits of endianess-related byte shuffling... or is it simply the "nature of the beast" that when working on big-endian / bi-endian systems, the `Target` author will _need_ to be keeping endianess in mind, even in their `Target` specific code?

If its the latter, maybe `gdbstub`'s stance is "when it comes to endianess, you're on your own." But if there's a chance the former is true... the `gdbstub` ethos is to always make things as easy for end users as possible, even if it means complicating some of the `gdbstub` internals to do so.

xobs, 2025-05-24:
I discovered why `probe-rs` is fine when it comes to memory -- they use `.read()` and `.write_8()`, both of which access target memory in an unordered manner. So when GDB sees that, it translates the addresses on its own. If it used their `.read_32()` and `.write_32()` instead, then the bytes would get swapped on the trip from the target to `probe-rs`, and would get shipped off to gdb in a reversed manner.

So as long as you shuffle the data from the target to gdb unordered, you're fine.

The problem with `probe-rs` is that register values come from the ADIv5 debug bridge, which is in LE, so register values are always in LE format. Which is why we need to swap registers when they get sent to gdb.

I'm unclear how much of an issue this is. For example, with the Xous debugger it's all native, so there wouldn't be any register swapping to do.