daniel5151/gdbstub #27: 0.2.2 semantic version is incompatible

h33p, 2020-09-11:
Several methods in `Target` trait break semantic versioning upon updating to version `0.2.2`, such as `read_addrs`, `write_addrs`, and `resume`.

This leads to weird bugs with crates using `0.2` version of gdbstub compiling just fine with existing Cargo.lock file, but failing when running `cargo package`.

I'm not sure what would be the best move here, but [cargo docs would suggest bumping the minor version number](https://doc.rust-lang.org/cargo/reference/manifest.html#the-version-field). I suppose that in addition, yanking `0.2.2` would make it ideal.

daniel5151, 2020-09-11:
Ahh, I didn't realize that cargo had stricter semver requirements than the official semver spec!

I was going off of point 4 from the semver spec: https://semver.org/#spec-item-4

> Major version zero (0.y.z) is for initial development. Anything MAY change at any time. The public API SHOULD NOT be considered stable.

A shame, since I've got a big API overhaul in the works which is more "worthy" of the `0.3` moniker...

It'll require some git-fu, but I'll go ahead and yank `0.2.2` and push up `0.3.0` instead.

daniel5151, 2020-09-11:
Alrighty, I've yanked `0.2.2` and pushed up `0.3.0`. Thanks for the heads up!

closed 2020-09-11 by daniel5151