daniel5151/gdbstub #28: Set `TCP_NODELAY` in examples and documentation

mchesser, 2020-09-15:
Since the protocol was originally designed to run over serial connections, it involves sending very small packets (i.e. the single byte step packet) and immediately waiting for the response. This behaves badly with the default TCP buffering settings (i.e. Nagle's algorithm). Causing it (at least on some platforms) to have an unusable amount of latency. 

Most gdbserver implementations I've seen set the `TCP_NODELAY` option to avoid this buffering. You can see this being done in gdbserver here:

https://github.com/bminor/binutils-gdb/blob/aea44f64c827932a2c67aeb2ee35a332696df8e8/gdbserver/remote-utils.cc#L155-L159

It might be useful for the documentation/examples to reference this, since I remember it being some what difficult to diagnose when I initially came across it (and I temporarily forgot about it when updating my code to use this crate since I was following the examples).

daniel5151, 2020-09-15:
Ah, interesting!

Poking around the internet, it looks like some TCP stacks will automatically disable Nagle's algorithm for loopback traffic, and since I've never actually tested `gdbstub` over a proper network connection, this issue's never come up. Neato

Looks like `TcpStream` has a [`set_nodelay`](https://doc.rust-lang.org/std/net/struct.TcpStream.html#method.set_nodelay) method, so this should be a quick and easy fix.

Thanks for the heads up!

closed 2020-09-15 by daniel5151

daniel5151, 2020-09-15:
Upon further consideration, instead of just documenting this gotcha, I've decided to fix the issue at the source level.

Since `0.4` is already bringing a bunch of API changes, I've gone ahead and added a new `on_session_start` method to the `Connection` trait, which is called prior to sending / receiving any GDB packets. It's includes a default no-op implementation, and is explicitly marked as optional.

With this new API, we can nip the issue at the source:

```rust
impl Connection for TcpStream {
    fn on_session_start(&mut self) -> Result<(), Self::Error> {
        // see issue #28
        self.set_nodelay(true)
    }
}

```

Nice!

* * *

By the way, after implementing this change, I saw a _noticeable_ improvement in debugging performance over loopback TCP.  It was by no means "unusable" beforehand, but it's now a lot snappier.

Thanks again!

commit f9598e99 (2020-09-15): add Connection::on_session_start + set_nodelay on TcpStream

Fixes #28

And wow, that really improved performance over loopback TCP.