daniel5151/gdbstub #29: Implement missing arch-specific `RegId` implementations

daniel5151, 2020-09-17:
# Overview

#22 added support for register-level read/writes, and introduced a new `RegId` associated type to the existing `Registers` trait. This associated type is used to translate raw GDB register ids (i.e: a arch-dependent `usize`) into a structured human-readable enum identifying the register.

e.g:

```rust
/// 32-bit ARM core register identifier.
#[derive(Debug, Clone, Copy)]
pub enum ArmCoreRegId {
    /// General purpose registers (R0-R12)
    Gpr(u8),
    /// Stack Pointer (R13)
    Sp,
    /// Link Register (R14)
    Lr,
    /// Program Counter (R15)
    Pc,
    /// Floating point registers (F0-F7)
    Fpr(u8),
    /// Floating point status
    Fps,
    /// Current Program Status Register (cpsr)
    Cpsr,
}

impl RegId for ArmCoreRegId {
    fn from_raw_id(id: usize) -> Option<(Self, usize)> {
        match id {
            0..=12 => Some((Self::Gpr(id as u8), 4)),
            13 => Some((Self::Sp, 4)),
            14 => Some((Self::Lr, 4)),
            15 => Some((Self::Pc, 4)),
            16..=23 => Some((Self::Fpr(id as u8), 4)),
            25 => Some((Self::Cpsr, 4)),
            _ => None,
        }
    }
}
```

Unfortunately, this feature was only implemented _after_ several contributors had already upstreamed their `Arch` implementations. 
As such, the `RawRegId` type was implemented as a "stop gap" measure to allow users to manually implement `read/write_registers`, even if it meant looking up the raw GDB register numbers themselves.

Please refer to the docs for more details on `RawRegId`, but at a glance:

```rust
#[derive(Debug)]
pub struct RawRegId(pub usize);

impl RegId for RawRegId {
    // simply pass-through the raw register ID +
    fn from_raw_id(id: usize) -> Option<(Self, usize)> {
        Some((RawRegId(id), 256 / 8))
    }
}
```

# Action Items

While all future `arch` implementations will be required to implement a proper `RegId`, the older `arch` implementations will have to be manually updated to support the new `RegId` system.

At the moment, these are the `arch` implementations that are in the main repo.
Those with a [x] have a valid `RegId`, while those with a [ ] are still using `RawRegId`:

- arm
  - [x] `Armv4t`
- mips
  - [ ] `Mips` / `Mips64`
- msp430
  - [ ] `Msp430`
- ppc
  - [ ] `PowerPcAltivec32`
- riscv
  - [x] `Riscv32` / `Riscv64`
- x86
  - [ ] x86 (i386)
  - [ ] x86_64

DrChat, 2021-05-27:
Here is the relevant target description file from the GDB sources for PPC32:
https://github.com/bminor/binutils-gdb/blob/master/gdb/features/rs6000/powerpc-32.c

daniel5151, 2021-05-27:
I would also leave the following links as well:

- https://github.com/bminor/binutils-gdb/blob/master/gdb/features/rs6000/powerpc-32.xml
- https://github.com/bminor/binutils-gdb/blob/master/gdb/features/rs6000/power-core.xml
- https://github.com/bminor/binutils-gdb/blob/master/gdb/features/rs6000/power-fpu.xml

Just as a reminder though, this issue is not a bug or a blocker - it is simply tracking something that should get resolved _eventually_. Moreover, implementing a new `RegId` isn't just adding a new `impl RegId` enum to the codebase -- you'd also have to put in some legwork to actually _test_ the implementation as part of a larger project.

@jamcleod, you wouldn't happen to have some spare time on your hands to circle back and upstream a `RegId` implementation for `PowerPcAltivec32`? I know it's been a while, but I thought I'd ask since you were the one who originally upstreamed the rest of the PowerPC code.

If not, all good! I totally don't mind leaving this issue open until someone happens to use `gdbstub` with a Power PC target, and is willing to upstream a tested `RegId` implementation :)

DrChat, 2021-05-27:
Appreciate the other references :)

I have a PPC target, but I'll have to figure out the dynamic dispatch first before I can test a `RegId` implementation for it.

daniel5151, 2021-05-27:
Ah, well that's great then!

In that case, I'm looking forwards to seeing a PR 😄