daniel5151/gdbstub #31: [Feedback] Overall API design

daniel5151, 2020-10-03:
Did you end up using `gdbstub` in your project? 
Leave a comment letting me know how the integration went!

- If everything went smoothly, awesome! That's makes me more confident leaving the API as-is on the road to `1.0.0`.
- Hit a pain point? Let me know how you think it could be improved!

**Please don't comment about missing protocol features or bugs.** Those should be filed as separate issues.

* * *

This issues will remain open until `gdbstub` hits version `1.0.0`.

DrChat, 2021-03-10:
Overall the API is a delight and painless to use. I'm using this library as a frontend to an emulator and learning Rust at the same time, and this is a breeze compared to our old solution.

One tidbit I'm running into now is that the current implementation for handling registers is mostly static - i.e. the target is not passed into `RegId` or `Registers` for handling any dynamic state (such as different register sets being available depending on the CPU mode, or the CPU architecture not being known until runtime).
Perhaps it should be rethought a bit so that it can be a bit more dynamic (but not necessarily dynamic enough that the return codes will differ across two sequential invocations on those traits).

DrChat, 2021-03-11:
I appreciate the detailed write-up! I'll have to take a look at implementing a passthrough like you described.

You're right that it's not ideal for users to know the layout of any GDB internals, and the current API is definitely an improvement over the old one.

Though, I just had a thought: Could we define the register file for each architecture using a `#[repr(C)]` struct (with a module such as [zerocopy::byteorder](https://docs.rs/zerocopy/0.4.0/zerocopy/byteorder/index.html) for endianness), and use a small unsafe block to cast the raw bytes from GDB into the register file struct?
Doing so could allow us to just eliminate `gdb_serialize` and `gdb_deserialize` altogether. For a passthrough, we'd just set the type of `Registers` to something like `[u8]` (something that would make the cast a no-op), and then in our generic architecture we would cast the register file depending on what architecture we're running.

Something like:
```
            ext::Base::G(cmd) => {
                let regs: <T::Arch as Arch>::Registers = unsafe {
                    // error checks, etc...
                    cmd.vals.align_to()
                };

                match target.base_ops() {
                    BaseOps::SingleThread(ops) => ops.write_registers(&regs),
                    BaseOps::MultiThread(ops) => ops.write_registers(&regs, self.current_mem_tid),
                }
                .handle_error()?;

                HandlerStatus::NeedsOK
            }
```

daniel5151, 2021-03-12:
To preface, here is a brief overview of the current `read/write_registers` implementation:

```
(alloc A) GDB Packet (ASCII, hex encoded, Big Endian) (in GDB-specific order)
Size: N

    <-- hex-en/decoded in-place --> 

(alloc A) GDB Packet (Binary, Big Endian) (in GDB-specific order)
Size: N/2 (each byte takes 2 ASCII bytes)

    <-- gdb_de/serialize --> 

(alloc B) Arch::Registers (arbitrary user-defined Rust struct, Machine Endian - almost always Little Endian) 
Size: >N/2 (not fully N, but likely not #[repr(packed)] as to be N/2)

    <-- read/write_registers --> 

(alloc C / read from registers directly) Target-specific code
```

Theoretically, your proposal to use zerocopy::byteorder would indeed work, and enable an "automatic" implementation of `gdb_de/serialize` by leveraging `#[repr(C, packed)]` + ensuring that the elements of the struct match the GDB order. Plus, you'd be avoiding that extra `alloc B` allocation. These are both good pros!

Unfortunately, this approach does come with a few cons:

- I'm not sure how `zerocopy` (or other similar-crates) would interact with `#[repr(packed)]`.
- While you save the extra allocation, you'll still end up paying the cost of swizzling the byteorder when reading/writing on little-endian platforms (i.e: the vast majority of platforms).
- The GDB ordering can be a bit... illogical at times, and tying it directly to the `Registers` type is a bit unfortunate. e.g: see the ARM core registers "floating point registers"
-  (arguably most importantly) it would require adding a nugget of `unsafe` code to `gdbstub`, which is something I've [somewhat miraculously] avoided up until now!

Now, none of these cons are deal-breakers, but they do make me wonder whether or not saving the extra copy is worth it.

If you want to keep riffing on this idea, how about you open a new issue and link back to these comments?
Also, if you have the time for it and care strongly about that extra allocation, feel free to take a crack at hacking together a working implementation - I would love to see a PR for it!

DrChat, 2021-03-16:
Ah - you have some really good points! I'm still learning a lot about the language, and it's easy to think "ah, well I can just do it the way I would've done it in C" :)

The extra copy and serialize/deserialize isn't that important for me to eliminate (as of now), so I probably won't pursue it too much further. My thought process was moreso to have the ability to directly reuse the existing register definitions in a generic architecture, and on further research it looks like that is do-able using the passthrough method you described.

---
There are a couple of other issues though that need further research, namely that `RegId::from_raw_id()` and `Arch::target_description_xml()` are called statically and can't do dynamic dispatch based on the running architecture. It doesn't look like there's a way to pass these values through to the `Target` either.
For `RegId`, the GDB ID can simply be passed through to `[read/write]_register`, but `RegId::from_raw_id()` also returns a register size parameter, which can vary depending on the target architecture.
`Arch::target_description_xml()` will need to be handled differently since it does not interact with functionality implemented on `Target`.

I wonder if there's a way to support both the current static implementation and a more dynamic way (should a `Target` want to do so).

daniel5151, 2021-03-16:
> but `RegId::from_raw_id()` also returns a register size parameter, which can vary depending on the target architecture

Ah, this is a very fair point, since the register size parameter determines the size of the buffer sent back to the GDB client...

One quick (albeit breaking) fix would be to change `read_register` to have a return type of `TargetResult<usize, Self>`, where `usize` is the number of bytes written into the `dst` buffer. This would mirror the standard library's `Read::read` method, and enable writing a "passthrough" `RegId` implementation that uses a sufficiently-large register size value to fit (almost) any arch's registers. A bit janky, but it should work.

> `Arch::target_description_xml()` [is] called statically and can't do dynamic dispatch based on the running architecture.

On a broader note, I've been meaning to overhaul and rework `Arch::target_description_xml()` for other reasons as well (tracked under #12). If possible, it'd be nice to hit two birds with one stone while doing other `Arch` related breaking changes.

* * *

Thanks for bringing up these valuable points about runtime configurable `Arch` implementations, I really appreciate it! Heck, I'd actually planned to use `gdbstub` in a multi-system emulator myself at some point, so it's good that we're catching and ironing out these issues now :smile: 

I'm excited to see what you come up with, and if you need any help / want to spitball some ideas, don't hesitate to open it as a draft and get some early feedback.

DrChat, 2021-04-13:
Another thing that recently came up - the implementation of `resume` for `MultiThreadOps` (and this may apply to `SingleThreadOps`) is somewhat easy to implement incorrectly.
It's provided an iterable list `Actions`, so my assumption was that I should iterate through that list and execute all requested actions.

However, when single-stepping a core, the GDB client will request a single-step on the target core _and_ a continue on all other cores.
This means that the GDB client wants the target core to single-step, and allow all other cores to execute while this operation is running.
I iterated through the list of actions in order, so I saw a single-step action followed by a continue action, which causes the single-step operation to simply become a continue operation.

The interface may need some rethinking. Per the GDB protocol, the list of actions are processed from left-to-right, wherein the leftmost action overrides actions specified later.

daniel5151, 2021-04-13:
Heck, I don't even provide a proper example of how to work with the current `Actions` based API in-tree (the `armv4t_multicore` example is _pretty bad_), so I'm not surprised it's easy to misuse 😅 

But yes, the current `resume` API is very easy to misuse, which is why I've been working on a pretty substantial resume API overhaul as part of `gdbstub 0.5`!
See https://github.com/daniel5151/gdbstub/blob/dev/0.5/src/target/ext/base/multithread.rs#L16-L106 as a starting point.

If at all possible, it'd be great if you could give this new API a shot. It's lumped in as part of some [other upcoming changes](https://github.com/daniel5151/gdbstub/blob/dev/0.5/CHANGELOG.md#050-dev), but they're mostly superficial code reorganization and/or internal refactoring, so it shouldn't be _too_ hard to upgrade from `0.4`. 

As always, any and all suggestions are much appreciated.

xobs, 2021-05-24:
I'm looking to add gdb support to my kernel by hooking an interrupt. Currently, the interrupt handler for the UART supports several single-key sequences to do things like dump the process table, count memory usage, and print process memory maps.

I would like to add a new single-key sequence `g` to enter gdbstub mode. Once this happens, all UART characters would be forwarded to `gdbstub`.

From what I understand, the `gdbstub` API is blocking. That is, I create a `GdbStub` struct with `GdbStub::new(...)`, then run it with `stub.run()`. This function then blocks until the remote client disconnects.

This doesn't work inside an interrupt handler.

What I'd rather have is a state machine into which I can pass characters. As this state machine operates, it calls various functions on the `Target`.

It almost looks like I could do this by turning the `recv_packet()` call inside-out. Another option may be to add separate functions such as `.start()` / `.pump(u8)` / `terminate()` that replace the generic `run()` call.

daniel5151, 2021-05-24:
Thanks for reaching out @xobs! 

I'm really glad you did, since validating `gdbstub`'s bare-metal use case is something I've been meaning to do for a [_very_ long time](https://github.com/daniel5151/gdbstub/commit/7daf21bc5930ca882596ca4736183e69d2b3f490#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R159). I too have a little hobby OS I work on from time to time, and one of the projects in my ever growing backlog is to add `gdbstub` support to it 😄 

Now, before responding with some preliminary thoughts on how to get `gdbstub` up and running in a kernel context, I should mention that if this is something you'd like to seriously pursue, I would be more than happy to work with you to get things up and running, and/or work on a PR to tweak `gdbstub`'s API to be more amenable to these kinds of use cases.

In addition, instead of cluttering this general "feedback" issue with our discussion, how about you open a dedicated tracking issue? Maybe something along the lines of "Validate bare-metal `no_std` `gdbstub` integration" (we can workshop the title later).

* * *

I've given it some thought, and I think that your state machine approach to reading packets may very well work!

I'll be honest, I never even considered using a state machine to ingest packets, since in the back of my mind, I was always planning on just spinning up a dedicated `GdbStub` kernel thread and re-using the existing blocking `Connection` abstraction. Your approach seems a lot leaner and easier to integrate thought!

Diving into the implementation details, it seems that I only ever call `Connection::read` within `recv_packet`, so it should be entirely possible to provide an alternative `GdbStub` implementation / set of methods that uses a "pump" based interface instead. There will be some subtleties around how the API would actually look / work under-the-hood, but we can sort out those details in a separate thread...

Another thing to keep in mind is how to handle GDB client interrupts. `gdbstub` currently provides a poll-based mechanism to check for interrupts (via [`GdbInterrupt`](https://docs.rs/gdbstub/0.5.0/gdbstub/target/ext/base/struct.GdbInterrupt.html)), which relies on the `Connection::peek` method under-the-hood to check if any breakpoint packets have arrived. In a interrupt-driven kernel debugging context, you could probably just bypass this poll-based mechanism entirely, and instead have your `resume()` handler work alongside your interrupt handler to return a `StopReason::GdbInterrupt` if a breakpoint byte comes down the wire while in GDB gdbstub mode.

* * *

Let me know if that answered your question! Like I mentioned above, if you want to keep this convo going, lets continue in a dedicated tracking issue.