daniel5151/gdbstub #35: [arch][x86] Break GPRs & SRs into individual fields/variants

daniel5151, 2020-10-29:
At the moment, the `X86_64CoreRegs::regs` field and `X86[_64]CoreRegs::segments` fields rely on an implicit ordering documented in their respective comments. It would be much better if explicit fields were used instead. 

As pointed out in https://github.com/daniel5151/gdbstub/pull/34#discussion_r513264251, there is no "one true" ordering for these registers, which could result in some nasty bugs where end-users de/serialize registers in the wrong order.

The corresponding `X86[_64]CoreRegId` enums would have to be changed as well.

* * *

This would be a breaking change.

daniel5151, 2021-05-02:
Removing the 0.5 milestone, now that #45 has been resolved.
That said, I might still implement this change as part of `gdbstub_arch`'s first release.

mchesser, 2021-05-05:
Not sure if it belongs in this issue, but it'd be nice to have the `IA32_FS_BASE`, `IA32_GS_BASE` MSRs exposed (used for TLS), exposed in GDB as `fs_base` and `gs_base` (https://github.com/bminor/binutils-gdb/blob/master/gdb/features/i386/64bit-segments.xml).

daniel5151, 2021-05-05:
@mchesser I've got some thoughts on that, but I'd rather write them down in a separate issue. Could you file one?

commit f1ea387c (2021-05-03): break up x86 segment registers into individual fields

closes #35
closed 2021-05-22 by daniel5151