daniel5151/gdbstub #41: `output!` macro in console_output.rs uses `writeln!` instead of intended `write!`

DrChat, 2021-03-06:
https://github.com/daniel5151/gdbstub/blob/bfe9a7b92ef86f2638c7207fb7d561e30432f85f/src/protocol/console_output.rs#L74-L80

This causes newlines to be printed to the GDB console when using the `output!` macro.

daniel5151, 2021-03-06:
Good catch!

I'll try to find some time this weekend to publish `gdbstub 0.4.4` that fixes this issue.

Thanks for reporting it!

DrChat, 2021-03-06:
Sounds good - appreciate it!

daniel5151, 2021-03-07:
The fix is in, and included in the just-published `gdbstub 0.4.4`!

closed 2021-03-07 by daniel5151

DrChat, 2021-03-07:
Appreciate the quick turnaround!

daniel5151, 2021-03-07:
Yeah, no problem! 
It's not like the project is active and popular enough to require rolling releases, so I may as well push out fixes as they come in 😄 

* * *

By the way, if you get the chance, let me know what you thought about `gdbstub` by dropping a comment on #31! I know the library's structure is a bit... unorthadox compared to other Rust libraries, so any and all feedback about the API would be much appreciated!

commit 59b180e0 (2021-03-07): use `write!` instead of `writeln!` in `output!`

closes #41