daniel5151/gdbstub #49: [RFC] s/managed/bytes?

wangbj, 2021-04-27:
`bytes` create is widely used with networking code. It could bend quite well with `Packet` and `Command` in `gdbstub`. `bytes` is actively maintained (compared to `managed` crate which doesn't have much commits in about two years), supports `no_std` (>=0.5.5), and zero copy operations; with `bytes::BufMut`, you can also `push<value: T>` with `bytes`, so no need to have `util/managed_vec.rs`. Most importantly, since it is isomorphic to `Arc<Vec<u8>>`, using `bytes` allows user to get rid of the annoying lifetime annotations (with slices), which can be found in many places in `src/protocol`. What do you think?

daniel5151, 2021-04-27:
Hey, thanks for bringing this up!

One of `gdbstub`'s use-cases is to provide a zero-_allocation_ implementation of the GDB Remote Serial Protocol, such that it can be used on embedded platforms that don't have a global allocator.

Unfortunately, while `bytes` is indeed `no_std`, it is _not_ "no_alloc", and requires a global allocator (relevant import [here](https://github.com/tokio-rs/bytes/blob/7b18c1c/src/lib.rs#L79)). 

And for the record, I don't think the lifetimes in `protocol` are that tricky to work with (though you could say that I'm somewhat biased 😄)

> compared to `managed` crate which doesn't have much commits in about two years

If you look under the hood of the `managed` crate, you'll find there isn't much code that needs iterating on 😉 

If you're worried about having `managed` as a dependency, I wouldn't be opposed to a PR that copied the `managed` code in-tree (which is totally fair-game thanks to the library being licensed as [BSD0](https://github.com/smoltcp-rs/rust-managed/blob/master/LICENSE-0BSD.txt)).

* * *

If that answers your quesiton, feel free to close this issue.

wangbj, 2021-04-27:
Thanks for the prompt reply, I'm closing this issue.

closed 2021-04-27 by wangbj