daniel5151/gdbstub #55: Add Sega Genesis support [FIX NOW]

jonahdlin, 2021-05-23:
I'm incredibly disappointed.

As a long time user of gdbstub I was shocked to see there is absolutely no support for the Sega Genesis. Just the other day I attempted to debug my play session of [Disney's Aladdin for the Genesis](https://en.wikipedia.org/wiki/Tekken_7) and thought I was in an Orwellian nightmare when I found that not only could gdbstub not set breakpoints within the Aladdin game, but it failed to fulfill its API for any game on the Sega Genesis I tried it on.

Do you think you're so high and mighty as to exclude the Sega Genesis from the catalogue of supported hardware? That console could eat you for breakfast, lunch and dinner and still have room for dessert. That console could grind this paltry repo into a fine powder, rim a fine glass of 1998 Veuve Clicquot La Grande Dame with it and enjoy it to the crisp, pulchritudinous sounds of Street Fighter II: Champion Edition.

You can do better. As the maintainer of this library I am positively floored that you could be so inept. People like myself are counting on you to at least approximate competence. Yes, the library is open-source and yes, the library is free. Let me tell you something pal, have you heard of **_exposure_**? I've starred this on GitHub. _Starred_. Well sir after this gross exploitation of the sacred trust between open-source user and open-source maintainer I can assure you I will be revoking the star I gave you with gusto.

daniel5151, 2021-05-23:
Thank you for bringing this critical omission to light.

Based on some cursory research (you'll have to apologize - I'm clearly an uncultured programmer who lacks experience working with such an ikonic console 🎮), it seems the Sega Genesis is based around a Motorola 68000 (m68k) CPU, and as such, it should be relatively straightforward to add support for it in `gdbstub` via a custom [`Arch`](https://docs.rs/gdbstub/0.5.0/gdbstub/arch/trait.Arch.html) trait implementation.

The relevant `target.xml` file can be found here: https://github.com/bminor/binutils-gdb/blob/master/gdb/features/m68k-core.xml

If you are interested in seeing a m86k target upstreamed in `gdbstub`, I would be more than glad to upstream any PRs sent my way.

Again, I sincerely apologize for omitting this critical architecture from `gdbstub_arch`, and I do hope this issue can be closed in a timely manner.

daniel5151, 2023-11-22:
This is good meme (❤️), but i'm closing it out for boring serious reasons. I should probably make it clear that requesting arch implementations as issues (without an associated implementation PR) isn't something folks should be doing.

closed 2023-11-22 by daniel5151