daniel5151/gdbstub #59: Make single-stepping optional

daniel5151, 2021-06-06:
A close reading of the GDB RSP docs suggests that single-stepping is actually an _optional_ feature of the protocol. Emphasis mine:
https://sourceware.org/gdb/current/onlinedocs/gdb/Overview.html#Overview

> At a minimum, a stub is required to support the ‘?’ command to tell GDB the reason for halting, ‘g’ and ‘G’ commands for register access, and the ‘m’ and ‘M’ commands for memory access. Stubs that only control single-threaded targets can implement run control with the ‘c’ (continue) command, **and if the target architecture supports hardware-assisted single-stepping, the ‘s’ (step) command**. Stubs that support multi-threading targets should support the ‘vCont’ command. All other commands are optional.

At first blush, it would seem that making single-stepping an optional feature is actually fairly straightforward: simply tweak the `resume` API to model single-stepping in a manner similar to other optional resume modes (such as optimized range stepping). It would require an API breaking change (removing the `ResumeReason::Step{WithSignal}` variants), but aside from that, it shouldn't be too tricky to implement.

* * *

Unfortunately, making single-stepping optional comes with a hidden footgun: if a GDB user tries to single-step on a target that don't support native single-stepping, the GDB client will instead attempt to use temporary breakpoints to "emulate" single stepping. You can see this behavior by skimming through the source of [`gdb/infrun.c:resume_1`](https://github.com/bminor/binutils-gdb/blob/aeeb758df5a61a323de0095c28b3de1ee7b00db3/gdb/infrun.c#L2188).

The `gdbstub` documentation should make it _very clear_ that supporting optimized single stepping is _highly recommended_, as it is significantly more efficient than relying on continue + temporary breakpoints. Moreover, if the target doesn't support single stepping _or_ breakpoints, then calling `step` won't work at all.

bet4it, 2021-10-04:
`resume` could also be optional?

daniel5151, 2021-10-04:
As per https://sourceware.org/gdb/onlinedocs/gdb/Overview.html#Overview

> At a minimum, a stub is required to support the ‘?’ command to tell GDB the reason for halting, ‘g’ and ‘G’ commands for register access, and the ‘m’ and ‘M’ commands for memory access. **Stubs that [...]

While it's technically possible to start a debugging session without implementing `resume`, the spec still mandates that resumption must be supported by the target.

If the target doesn't wish to implement `resume`, it can simply provide a dummy implementation, such as immediately returning a `StopReason::Signal(5)`.

bet4it, 2021-10-04:
> Stubs that only control single-threaded targets **can implement run control** with the ‘c’ (continue) command

It's just 'can', not 'must' or 'should', right?

> Stubs that support multi-threading targets should support the ‘vCont’ command

It's talking about `vCont`, but `vCont` has some subcommands such as `vCont;c` or `vCont;s`.
For example, I may only support `vCont;s` and `vCont;S`, I don't think I violate this spec.

* * *

The document here is talking about "run control", but is run control necessary? I can think of a situation that I just want to use GDB RSP to probe current memory and register status, then I will 'detach' rather than 'continue'.

* * *
> If the target doesn't wish to implement `resume`, it can simply provide a dummy implementation, such as immediately returning a `StopReason::Signal(5)`.

We can also do the same thing to `step`, so why you make it optional?

daniel5151, 2021-10-04:
Ahh, you know what, that's totally a reasonable way to read the spec.

In that case, it might even make sense to spit the entire notion of "resumption" into a separate IDET, leaving BaseOps to only deal with reading/writing registers/memory + thread enumeration... 

While that would technically be more "correct" with the base spec, this might be a good moment to discuss the entire concept of "base spec" in `gdbstub`:

* * *

As hypocritical as it sounds coming from me (given my proclivity to follow the spec to a tee), `gdbstub` has always taken a more "pragmatic" approach to what constitutes the GDB RSP's "base" protocol.

A strict reading of the spec would suggest that the only packets that need to be implemented by a minimal stub are _literally_ `?`, `m`, `M`, `g`, and `G`, but in reality, trying to connect to a GDB stub that only supports those packets using a "modern" GDB client will fail miserably. Namely, a minimal stub must _also_ support packets such as `H`, `qAttached`, `qsThreadInfo`, `qSupported`, `T`, etc... to get past the initial handshake.

In addition, in order to simplify `gdbstub`'s internal architecture, `gdbstub` reports support for the GDB RSP's multiprocess extensions (i.e: `vCont`, expanded Thread Id syntax, etc...). Theoretically, this is something I _might_ be able to make optional, but it would be quite tricky, as the way `ThreadId` structs are parsed would depend on if the IDET is enabled, which might be tricky to "plumb through". Not to mention, I suspect that several newer protocol extensions would simply _assume_ multiprocess extensions are enabled, and would subtly not work with a stub that doesn't support them...

Lastly, `gdbstub` also unconditionally implements several "end of connection" packets, such as `D`, `k`, and `vKill` to support graceful exit from a `gdbstub` debug session. Though now that I think about it, these could _also_ be made optional...

Long story short: when it comes to the "base" protocol (i.e: the extensions that _most_ implementations would be expected to implement), `gdbstub` has typically taken a more pragmatic approach, whereby it _does_ consider the fact that most users will be using the standard GDB client, and that said client doesn't necessarily respect the altruistically optional nature of a minimal GDB RSP implementation.

* * *

With this in mind, how does this "pragmatic" approach relate to how `gdbstub` handles `resume`?

You're _technically_ correct that - at least according to the spec - it _should_ be possible to implement a stub that only supports single-stepping, while _not_ supporting continue (i.e: by reporting back only `vCont;s;S`). Unfortunately, if we peek under the hood of the GDB Client implementation, you'll find [this code](https://github.com/bminor/binutils-gdb/blob/aeeb758df5a61a323de0095c28b3de1ee7b00db3/gdb/remote.c#L6150-L6153), which effectively _enforces_ that clients support continue. If you poke around the code some more, you'll also find that the GDB client assumes that `c` is also unconditionally supported.

In other words: the mainline GDB client _cannot_ drive a GDB RSP target using only single-stepping, and assumes that the target supports continue.

IMHO, that seems like a pretty reasonable assumption. After all, it's far easier to implement continue that it is to implement single-stepping.

> > If the target doesn't wish to implement `resume`, it can simply provide a dummy implementation, such as immediately returning a `StopReason::Signal(5)`.
> 
> We can also do the same thing to `step`, so why you make it optional?

Consider the case where a client wants to support `continue`, but doesn't want to explicitly implement `step`. This is a use-case that I've come across myself recently.

With the current implementation (i.e: `step` support is _always_ reported to the GDB client), it is _invalid_ to simply return  `StopReason::Signal(5)` from the single-step operation, as the target wouldn't have actually single stepped, which could lead to Bad Things happening. Instead, a implementer is _forced_ to get single-stepping working as part of their initial `gdbstub` bringup, which doesn't need to be the case if `step` was optional.

* * *

TL;DR

- the entire notion of "resuming the target" could probably be put into a separate IDET, and `gdbstub`'s minimal configuration would therefore only allow connecting to a paused target + manipulating its memory/register state.
    - I'll actually seriously consider doing this.
- while `continue` is not required in the strict reading of the spec, it is de-factor required by the mainline GDB client implementation, and since `gdbstub` takes a pragmatic approach to what constitutes the "base" protocol (in the name of user empathy), `gdbstub` will require that users support `continue`.

bet4it, 2021-10-06:
> the entire notion of "resuming the target" could probably be put into a separate IDET, and `gdbstub`'s minimal configuration would therefore only allow connecting to a paused target + manipulating its memory/register state. 

That's what I want!
Then we should set `resume` to be optional, and rename it to `continue`.

* * *
> Unfortunately, if we peek under the hood of the GDB Client implementation, you'll find [this [...]

I think the "base" protocol we must implement includes commands GDB will request automatically when connect. If you capture the packets when connect GDB RSP server with GDB, you can find that `vCont?` won't be requested until you send commands such as `s` or `c`. I'm not sure if we can only implement single-step but not continue, but I'm sure that we can implement with neither of single-step nor continue is implemented.

If users not implement `resume`, they may encounter some errors when using some commands, but it's their responsibility to deal with it. For example, we set `step` to optional because if user not implement it, GDB will step manually with software breakpoint. But what if software breakpoint is also not implemented? We can't do all the things for the user.

And it's more friendly for newbies if we set `resume` to optional. It's much easier to implement register/memory access than run control. If a newbie try to use `gdbstub` on his repository, he can set up a GDB RSP server easily without need to change much code if run control is not considered.

daniel5151, 2021-10-06:
> I think the "base" protocol we must implement includes commands GDB will request automatically when connect.

The more I think about it, the more I think this is actually the right approach...

The only question is whether or not this should be a part of #92. Technically, the change of "making single stepping optional" and the change of "making resumption _in general_" option are two separate ones, and probably merit two separate PRs.

I think I'll open a new tracking issue to make resumption optional, and move forward with #92 in its current state.

> If users not implement `resume`, they may encounter some errors when using some commands, but it's their responsibility to deal with it. For example, we set `step` to optional because if user not [...]

Funny you bring this up as an example, because I literally added a guard rail for this lol: https://github.com/daniel5151/gdbstub/commit/1d55539b072bae4251becde8392b0b116a2e139c

daniel5151, 2021-10-08:
Yes, `continue` would absolutely be the most appropriate name - too bad it's a keyword 😅

Though I agree - in the single-step case, it might make sense to rename `resume` to something like `resume_cont` (and then likely change `step` to `resume_step` for consistency, etc...)

While this is technically bikeshedding, I do agree that the names ought to be clearer, so I'll update them in the current PR appropriately...

* * *

Aside from that though, it seems the specific issue of "making single step optional" has been pretty much resolved.

After tweaking some of the method names, I'll merge the current PR as is, and then look into splitting resumption as a whole out into an optional IDET in a follow up PR...

bet4it, 2021-10-08:
We could just use `cont`!