daniel5151/gdbstub #60: Pass kind argument to add_hw_watchpoint

bet4it, 2021-06-11:
Currently, when we add watchpoint, `kind` is ignored:
https://github.com/daniel5151/gdbstub/blob/a33bbaed77848ab6eefdedd0cb0bf4537ebb4c03/src/gdbstub_impl/ext/breakpoints.rs#L19-L32
But according to [gdb rsp doc](https://sourceware.org/gdb/onlinedocs/gdb/Packets.html#Packets), `kind` when set watchpoint means number of bytes to watch, which is not meaningless:
```
‘Z2,addr,kind’
    Insert (‘Z2’) or remove (‘z2’) a write watchpoint at addr. The number of bytes to watch is specified by kind. 
```
We need also pass `kind` as an argument to `add_hw_watchpoint`.

daniel5151, 2021-06-11:
Well, this is embarrassing 😳 
I'm surprised that I didn't catch this when I plumbed-through `BreakpointKind` support to software/hardware watchpoints... Oops.

In any case, thanks for reporting this discrepancy!
I'll land a fix in the `dev/0.6` branch that updates the `add_hw_watchpoint` function signature appropriately (hopefully sometime this weekend).

Since this will require an API breaking change, it will also require a breaking `gdbstub` release. That's not too bad, as there've been enough breaking changes trickling into the `dev/0.6` branch that I'll probably end up publishing `0.6` sometime relatively soon-ish. I don't have an exact timeline yet, but I'd wager `0.6` will be up on crates.io sometime in the next few weeks.

daniel5151, 2021-06-14:
I've pushed a fix to the `dev/0.6` branch. This issue will be closed once gdbstub 0.6 is published.

commit 62e76468 (2021-06-14): plumb watchpoint length to public API

closes #60

also took this chance to rewrite / optimize some of the code in the
breakpoint handler. notably, I added match arm guards to help with DCE.
closed 2021-09-24 by daniel5151