daniel5151/gdbstub #67: gdbstub_arch package doesn't contain LICENSE file

keiichiw, 2021-08-05:
`gdbstub_arch-0.1.0` package distributed on [crates.io](https://crates.io/crates/gdbstub_arch) doesn't contain LICENSE file while its Cargo.toml says its license is MIT.

```bash
$ curl -L 'https://crates.io/api/v1/crates/gdbstub_arch/0.1.0/download' > /tmp/gdbstub_arch-0.1.0.tar.gz
$ tar xvf /tmp/gdbstub_arch-0.1.0.tar.gz
$ cat gdbstub_arch-0.1.0/LICENSE
cat: gdbstub_arch-0.1.0/LICENSE: No such file or directory
```

@daniel5151 Could you release the updated version (0.1.1?) with LICENSE included?

daniel5151, 2021-08-05:
Ah, good catch! Sorry about that!

I'll publish a new version sometime today / tomorrow.

daniel5151, 2021-08-06:
`gdbstub_arch` 0.1.1 is now published, and includes a LICENSE file 🎉

closed 2021-08-06 by daniel5151

keiichiw, 2021-08-07:
Thank you for the quick update!

daniel5151, 2021-08-07:
No problem 😄 
And thank you for the support!