daniel5151/gdbstub #68: Relicense gdbstub under dual MIT/Apache-2.0

daniel5151, 2021-08-21:
Inspired by https://github.com/bevyengine/bevy/issues/2373 and https://github.com/bevyengine/bevy/pull/2509

When I released gdbstub 0.1, I didn't think too much about which license to use, and just picked one that seemed reasonable -  namely, the MIT license. Over time, as I've gotten more familiar with crate maintainership and open source licensing, I've come to realize that it'd probably be a good idea to switch over to the dual MIT/Apache-2.0 license that most of the Rust ecosystem uses.

For more context around the **Why** behind this relicense, check out https://github.com/bevyengine/bevy/issues/2373

This isn't urgent by any means, but I might as well get the ball rolling in case there are some contributors who are a bit slow to respond.

**If you are mentioned in this issue, I need your help to make this happen**

To agree to this relicense, please read the details in this issue, then leave a comment with the following message:

```
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.
```

If you disagree, please respond with your reasoning. Anyone who doesn't agree to the relicense will have any gdbstub contributions that qualify as "copyrightable" removed or re-implemented.

# What will this look like?

After getting explicit approval from each contributor of copyrightable work (as not all contributions qualify for copyright, due to not being a "creative work", e.g. a typo fix), we will do the following:

* Change the LICENSE file to describe the dual license. Move the MIT license to docs/LICENSE-MIT. Add the Apache-2.0 license to docs/LICENSE-APACHE
* Update the `gdbstub` and `gdbstub_arch` Cargo.toml files to use the new "MIT OR Apache-2.0" license value
* Add a License section to the main readme

Note: I do intend to keep the `Copyright (c) <year> Daniel Prilik` line in the licenses. Please let me know if you're strongly opposed to this.

# Contributor checkoff

- [ ] @bet4it 
- [x] @daniel5151 
- [ ] @DrChat 
- [ ] @fzyz999 
- [ ] @jamcleod
- [ ] @keiichiw 
- [ ] @mchesser 
- [ ] @sapir 
- [ ] @starfleetcadet75 
- [ ] @thomashk0 
- [ ] @Tiwalun 

# Contributors with "obsolete" changes 

(no need for approval)

- iburinoc

# Contributors with "trivial" changes that are ok to keep

(no need for approval)

- JamesHageman
- pheki

DrChat, 2021-08-21:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

👍

fzyz999, 2021-08-21:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

mchesser, 2021-08-21:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

Tiwalun, 2021-08-21:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

keiichiw, 2021-08-21:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

thomashk0, 2021-08-21:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

sapir, 2021-08-21:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

bet4it, 2021-08-22:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

jamcleod, 2021-08-28:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

(Sorry for taking so long to reply, github is very bad about notifying me!)

daniel5151, 2021-09-27:
pinging @starfleetcadet75, as it seems you're the last one who needs to sign off.

starfleetcadet75, 2021-11-09:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

Also sorry for taking so long

commit 0a40f163 (2021-11-16): relicense gdbstub as MIT OR Apache-2.0

closes #68
closed 2021-11-16 by daniel5151