daniel5151/gdbstub #74: deferred_stop_reason takes a ThreadStopReason in single-threaded mode?

gz, 2021-09-22:
I'm wondering if it's this a bug that the [deferred_stop_reason](https://github.com/daniel5151/gdbstub/blob/4344c8586b956493594eaa9d084b6b27f5c0f90a/src/gdbstub_impl/mod.rs#L377) takes a `crate::target::ext::base::multithread::ThreadStopReason` and is not parameterized to take a `gdbstub::target::ext::base::singlethread::StopReason` in case the gdbstub is for `BaseOps::SingleThread`?

daniel5151, 2021-09-22:
If you peek under-the-hood of `gdbstub`, you'll find that `singlethread::StopReason` gets converted into a `multithread::ThreadStopReason` with a constant [`SINGLE_THREAD_TID`](https://github.com/daniel5151/gdbstub/blob/4344c8586b956493594eaa9d084b6b27f5c0f90a/src/lib.rs#L180):

https://github.com/daniel5151/gdbstub/blob/4344c8586b956493594eaa9d084b6b27f5c0f90a/src/gdbstub_impl/ext/base.rs#L782-L804

It's been a while since I've hacked on this code directly, but IIRC I couldn't think of a great way to parameterize `deferred_stop_reason` over single/multi-threaded targets, so I took the "cop out" approach of simply using the more specific `multithread::ThreadStopReason` type.

That said, after reviewing the code with fresh eyes this morning, I suspect that long-term, I'll actually end up removing `singlethread::StopReason` entirely, and swapping all APIs over to using a single `StopReason` type which encodes both the TID, _and_ PID of the stopped thread (in anticipation of eventually supporting multi-_process_ debugging as well).

So, to answer your concrete question: yes, I'm aware of this weird aspect of the API, and I'm planning to solve it in one of two ways:

1. updating the documentation to make it _very_ clear how to use the state machine with single-threaded targets
2. making the point moot by removing "singlethread::StopReason" type entirely (along with any requisite doc changes)

gz, 2021-09-22:
Great, thanks for clarifying!

bet4it, 2021-09-25:
You could use `into`:
```rust
let reason = StopReason::DoneStep;
match gdb_inner.deferred_stop_reason(emu, reason.into())? {
    ...
}
```

gz, 2021-09-25:
agreed I didn't see the into() implementation. What was mainly confusing here for me was that in the single-threaded API there was a type from `crate::target::ext::base::multithread`. But I supposed that Daniels refactoring will probably move this type somewhere more general.

daniel5151, 2021-11-28:
This has been fixed in 2eca7335 and a24b7244

closed 2021-11-28 by daniel5151