daniel5151/gdbstub #80: RIP size on x86-64

gz, 2021-09-25:
[rip size](https://github.com/daniel5151/gdbstub/blob/52ae3c7cf2cb7298925a1ba5b4ed1e904ae64416/gdbstub_arch/src/x86/reg/id.rs#L175) is currently 4 bytes, but maybe it should be 8?

https://github.com/bminor/binutils-gdb/blob/master/gdb/features/i386/64bit-core.xml lists it as:

```xml
  <reg name="rip" bitsize="64" type="code_ptr"/>
```

bet4it, 2021-09-25:
I will fix it in a following PR ( with another way ) , wait for me😅

gz, 2021-09-25:
random thought: it might be nice to statically type-encode the size of the registers when when they're passed to the read/write_registers. One way could be to move the `val: &[u8]` and `dst: &mut [u8]` in some enum with the RegId and encode the length explicitly (specify as &[u8; 8] etc.)?

I'm thinking that might save some slice length checks when implementing these functions. But not sure if even possible just a thought :)

daniel5151, 2021-09-25:
@gz I assume you meant `read/write_register` (as opposed to `read/write_registers`)?

If so, then you might find this discussion from the initial implementation PR interesting: https://github.com/daniel5151/gdbstub/pull/22#issuecomment-683889998

Though now that I'm re-reading this idea a year later, I realize that there might be some easier approaches to pull off something similar... I certainly wouldn't be opposed to having stricter type-safety in these APIs, so I'm open to any ideas!

daniel5151, 2021-09-25:
And as for the original issue: yes, this seems like an oversight in the initial #34 implementation. 

Pinging @keiichiw, since this might translate to a subtle correctness issue in your implementation.

gz, 2021-09-25:
Yes I meant the single register writes with the dynamic `val` and `dst` slices.

daniel5151, 2021-09-27:
@gz if this is something you'd like fixed sooner, feel free to send a quick fixup PR. Otherwise, a fix for this will land alongside some other changes to `read_register` in #85

gz, 2021-09-27:
Sent pull request, yes would be nice to have this fixed, thanks!

PR #87 (2021-09-28): Change rip size to 8 in `X86_64CoreRegId`.
### API Stability

- [x] This PR does not require a breaking API change

Fixes #80.

### Checklist

I ran this in the past but this change was with the github editor :)
PR #85 (2021-09-29): Dynamic `read_register` + `RegId` support
### Description

New approach to #58.

Part of #70.

Closes #58. Closes #80.

### API Stability

- [ ] This PR does not require a breaking API change

### Checklist

- Implementation
  - [x] `cargo build` compiles without `errors` or `warnings`
  - [x] `cargo clippy` runs without `errors` or `warnings`
  - [x] `cargo fmt` was run
  - [x] All tests pass
- Documentation
  - [x] rustdoc + approprate inline code comments
  - [ ] Updated CHANGELOG.md
  - [ ] (if appropriate) Added feature to "Debugging Features" in README.md
- _If implementing a new protoco [... cut]
closed 2021-09-28 by daniel5151