daniel5151/gdbstub #81: Unsupported X packets (write binary data)

gz, 2021-09-25:
I'm getting some

```
22363003360 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("X40003cf7e750,0:")
```
lines in my gdb session.

Apparently X is for:

```
write mem (binary) | Xaddr,length:XX... | addr is address, length is number of bytes, XX... is binary data.  The characters $, #, and 0x7d are escaped using 0x7d.
```

If you can confirm that this is really missing (and not me doing something stupid again) I'll give it a shot and implement it.

daniel5151, 2021-09-25:
Yep, `gdbstub` doesn't currently implement `X`, since AFAIK, the GDB client simply falls back to using the `M` packet to write data to memory if `X` is not implemented. The `M` packet uses a less efficient ASCII hex representation of data, though it seems that no one had noticed or cared enough to implement the more efficient `X` packet.

Thankfully, this packet seems to be solely an "optimization" packet, and won't require a whole new user-facing API - it can be easily plumbed through via the existing `Target::write_addrs` APIs.

I'd be more than happy to merge an implementation of `X` packet handling, and would expect that the implementation be quite straightforward (i.e: it'd be a nearly exact mirror of the current `M` parsing code).

* * *

Also, sidenote: that example packet you gave seems a bit weird 👀
Why would GDB waste bandwidth trying to write data with a length of _zero_? 

Fascinating...

gz, 2021-09-25:
Yea not sure about that zero length packet. There is still some bugs in my implementation with stepping, either just getting stuck or reporting some errors like `Protocol error: P (set-register) conflicting enabled responses.` when I turn on single register writes. So can't tell yet if I broke something horribly or if it's really gdb doing that.

PR #82 (2021-09-27): Implement X packet for gdbstub.
### Description

Closes #81

### API Stability

- [X] This PR does not require a breaking API change

### Checklist

- Implementation
  - [X] `cargo build` compiles without `errors` or `warnings`
  - [X] `cargo clippy` runs without `errors` or `warnings`
  - [X] `cargo fmt` was run
  - [X] All tests pass
closed 2021-09-27 by daniel5151