daniel5151/gdbstub #83: Off-by-one register naming for xmm registers?

gz, 2021-09-25:
If I understand [this](https://github.com/bminor/binutils-gdb/blob/master/gdb/features/i386/64bit-sse.xml) correctly, then [line 180 and 181](https://github.com/daniel5151/gdbstub/blob/488a4b133a755c1009cca1d1560e3250fd456d31/gdbstub_arch/src/x86/reg/id.rs#L180) here should this instead say (e.g., currently only assumes 15 xmm regs):

```
            40..=56 => (Xmm(id as u8 - 40), 16),
            57 => (Mxcsr, 4),
```  
I found this by accident because my gdb sends a packet for setting reg 57/0x39 which is currently ignored (here with some added trace statements):
```
28275229934 [TRACE] - gdbstub::protocol::recv_packet: <-- $P39=ffffffffffffffff#59
28279335634 [INFO ] - gdbstub::gdbstub_impl::ext::single_register_access: got P P { reg_id: 57, val: [255, 255, 255, 255, 255, 255, 255, 255] }
28285262982 [INFO ] - gdbstub::gdbstub_impl::ext::single_register_access: reg is None
28289378772 [INFO ] - gdbstub::gdbstub_impl::ext::single_register_access: empty pkt?
28293528816 [TRACE] - gdbstub::protocol::response_writer: --> $#00
```

gz, 2021-09-25:
The 2nd mystery her is why gdb thinks sending 8 bytes is acceptable here because mxcsr is supposedly 32-bit.

```
  <reg name="mxcsr" bitsize="32" type="i386_mxcsr" group="vector"/>
```

daniel5151, 2021-09-25:
So... should this issue be closed as well?

gz, 2021-09-25:
well my only remaining issue is that I still can't really tell what register with ID 0x39 would be. It still wants to write to it but the register doesn't exist in the x86-64 arch in gdbstub.

gz, 2021-09-25:
To explain a bit more, I was trying to look at the [xml files](https://github.com/bminor/binutils-gdb/tree/master/gdb/features/i386) here. But they are somewhat confusing. I figured maybe 0x39 is some AVX register or something related since it's right after the xmm registers. But I couldn't really tell how these XML files get applied and in what order.

There is [this one](https://github.com/bminor/binutils-gdb/blob/master/gdb/features/i386/64bit-linux.xml#L10) which seems to be Linux specific?

gz, 2021-09-25:
Somewhat related: Without trace output it was hard to tell that the register wouldn't get set. Since the RegID was missing in `gdbstub_arch` for x86-64, gdbstub would just return to gdb which would eventually print `Protocol error: P (set-register) conflicting enabled responses.` (because we were able to set past registers with the function but not this one).

Maybe logging a `warn!` or `error!` if a write_register request comes in for a register that isn't known would be helpful for future users?

daniel5151, 2021-09-25:
Hmmm, interesting...

As a quick reminder: these built-in arch defns are all community contributions, so I've never actually done any hands-on validation of the x64 arch. That said, I'm more than happy to help you figure out what's going on here.

Could you repro the register write request, while also having debugging enabled on the GDB _client_ as well? I think the most relevant GDB commands would be:

```
set debug arch 1
set debug remote 1
set debug xml 1
set debug target 1
```

Also, what does `show arch` return after connecting to the target?

> Maybe logging a `warn!` or `error!` if a write_register request comes in for a register that isn't known would be helpful for future users?

That seems like a reasonable addition, though I'd say hold off on opening a PR for it - it's something I can just throw in with some other changes :)

gz, 2021-09-25:
gdbstub/log:

```
 840283670 [INFO ] - nrk::arch::gdb: Waiting for a GDB connection (I/O port 0x2f8)...                                       
 850291296 [INFO ] - nrk::arch::gdb: Use `target remote localhost:1234` in gdb to connect.                                  
4380516342 [INFO ] - nrk::arch::gdb: Debugger connected.                                                                    
4385886304 [WARN ] - nrk::arch::irq: Got debug interrupt Debug condition                                                    
4415342166 [TRACE] - gdbstub::protocol::recv_packet: <-- $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events
+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;xmlRegisters=i386#6a                                
4437018188 [TRACE] - gdbstub::protocol::response_writer: --> $PacketSize=1000;vContSupported+;multiprocess+;QStartNoAckMode+
;hwbreak+;qXfer:features:read+#f2                                                                                           
4444059074 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                  
4449171420 [TRACE] - gdbstub::protocol::recv_packet: <-- $vMustReplyEmpty#3a                                                
4453363374 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("vMustReplyEmpty")                                          
4457578078 [TRACE] - gdbstub::protocol::response_writer: --> $#00                                                           
4461711716 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                  
4466849212 [TRACE] - gdbstub::protocol::recv_packet: <-- $QStartNoAckMode#b0                                                
4471198514 [TRACE] - gdbstub::protocol::response_writer: --> $OK#9a                                                         
4475300722 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                  
4479334510 [TRACE] - gdbstub::protocol::recv_packet: <-- $Hgp0.0#ad                                                         
4483206904 [TRACE] - gdbstub::protocol::response_writer: --> $OK#9a                                                         
4580817552 [TRACE] - gdbstub::protocol::recv_packet: <-- $qXfer:features:read:target.xml:0,ffb#79                           
4591204068 [TRACE] - gdbstub::protocol::response_writer: --> $m<target version="1.0"><architecture>i386:x86-64</architecture
><feature name="org.gnu.gdb.i386.sse"></feature></target>#c7                                                                
4605889946 [TRACE] - gdbstub::protocol::recv_packet: <-- $qXfer:features:read:target.xml:76,ffb#b6                          
4612103508 [TRACE] - gdbstub::protocol::response_writer: --> $l#6c                                                          
4619367440 [TRACE] - gdbstub::protocol::recv_packet: <-- $qTStatus#49                                                       
4624229540 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("qTStatus")                                                 
4629359860 [TRACE] - gdbstub::protocol::response_writer: --> $#00                                                           
4725387364 [TRACE] - gdbstub::protocol::recv_packet: <-- $?#3f                                                              
4729103364 [TRACE] - gdbstub::protocol::response_writer: --> $S05#b8                                                        
4734787046 [TRACE] - gdbstub::protocol::recv_packet: <-- $qfThreadInfo#bb                                                   
4738904292 [TRACE] - gdbstub::protocol::response_writer: --> $mp01.01#cd                                                    
4744482530 [TRACE] - gdbstub::protocol::recv_packet: <-- $qsThreadInfo#c8                                                   
4748345052 [TRACE] - gdbstub::protocol::response_writer: --> $l#6c                                                          
4754153852 [TRACE] - gdbstub::protocol::recv_packet: <-- $qAttached:1#fa                                                    
4759208494 [TRACE] - gdbstub::protocol::response_writer: --> $1#31                                                          
4848619454 [TRACE] - gdbstub::protocol::recv_packet: <-- $Hc-1#09                                                           
4852407818 [TRACE] - gdbstub::protocol::response_writer: --> $OK#9a                                                         
4856826472 [TRACE] - gdbstub::protocol::recv_packet: <-- $qC#b4                                                             
4860423770 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("qC")                                                       
4864040356 [TRACE] - gdbstub::protocol::response_writer: --> $#00                                                           
4869162730 [TRACE] - gdbstub::protocol::recv_packet: <-- $qOffsets#4b                                                       
4874955834 [TRACE] - gdbstub::protocol::response_writer: --> $Text=40003cf2b000;Data=40003cf2b000;Bss=40003cf2b000#20       
4962649396 [TRACE] - gdbstub::protocol::recv_packet: <-- $g#67                                                              
4966262708 [TRACE] - nrk::arch::gdb: read_registers X86_64CoreRegs { regs: [0, 0, 70369772420144, 22, 70369771834825, 703697
[... cut ...]
remote:target_xfer_partial (2, (null), 0x0, 0x564a0dd00844, 0x40003d01dc3b, 1) = 1, 1, bytes = 48
<- remote->remove_breakpoint (i386:x86-64, 0x000040003d01dc3b, 0) = 0
-> remote->always_non_stop_p (...)
<- remote->always_non_stop_p () = false
-> remote->supports_terminal_ours (...)
<- remote->supports_terminal_ours () = false
Protocol error: P (set-register) conflicting enabled responses.
-> remote->terminal_save_inferior (...)
<- remote->terminal_save_inferior ()
-> remote->terminal_ours (...)
<- remote->terminal_ours ()
(gdb) 
```

daniel5151, 2021-09-26:
Ahh, it seems you've only enabled those debug logs _after_ connecting to the target. It'd be nice to see some of the initial handshake as well. You may need to modify your `.gdbinit` script to enabled debugging a bit earlier.

Also, some other ideas for eliminating some variables:

- Define a custom `Arch` implementation that uses a super-explicit `target.xml` implementation, spelling out exactly which registers ought to be available. A good starting point may be the x64 config that QEMU, though keep in mind you will need to tweak the implementations + layout of the `Registers` + `RegId` structs to match the XML file. https://github.com/qemu/qemu/blob/master/gdb-xml/i386-64bit.xml
- If you are able to boot your kernel in something like QEMU, see if you can trigger the same packet request.

gz, 2021-09-26:
Ah here is the log with debugging enabled in .gdbinit:

```
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ../target/x86_64-uefi/debug/esp/kernel...
-> exec->log_command (...)
<- exec->log_command (target remote localhost:1234)
-> remote->open (...)
target_close ()
Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;xmlRegisters=i386#6a...Ack
Packet received: PacketSize=1000;vContSupported+;multiprocess+;QStartNoAckMode+;hwbreak+;qXfer:features:read+
Packet qSupported (supported-packets) is supported
Sending packet: $vMustReplyEmpty#3a...Ack
Packet received: 
Sending packet: $QStartNoAckMode#b0...Ack
Packet received: OK
Sending packet: $Hgp0.0#ad...Packet received: OK
Sending packet: $qXfer:features:read:target.xml:0,ffb#79...Packet received: m<target version="1.0"><architecture>i386:x86-64</architecture><feature name="org.gnu.gdb.i386.sse"></feature></target>
remote:target_xfer_partial (10, target.xml, 0x556fd408fd90, 0x0, 0x0, 4096) = 1, 118, bytes =
 3c 74 61 72 67 65 74 20 76 65 72 73 69 6f 6e 3d ...
Sending packet: $qXfer:features:read:target.xml:76,ffb#b6...Packet received: l
remote:target_xfer_partial (10, target.xml, 0x556fd4233cb6, 0x0, 0x76, 4096) = 0, 0
target description (line 1): Starting:
<target version="1.0"><architecture>i386:x86-64</architecture><feature name="org.gnu.gdb.i386.sse"></feature></target>
target description (line 1): Entering element <target>
target description (line 1): Entering element <architecture>
target description (line 1): Leaving element <architecture>
target description (line 1): Entering element <feature>
target description (line 1): Leaving element <feature>
target description (line 1): Leaving element <target>
target description (line 1): XInclude processing succeeded.
target description (line 1): Starting:
<target version="1.0"><architecture>i386:x86-64</architecture><feature name="org.gnu.gdb.i386.sse"></feature></target>
target description (line 1): Entering element <target>
target description (line 1): Parsing attribute version="1.0"
target description (line 1): Entering element <architecture>
target description (line 1): Leaving element <architecture>
target description (line 1): Entering element <feature>
target description (line 1): Parsing attribute name="org.gnu.gdb.i386.sse"
target description (line 1): Leaving element <feature>
target description (line 1): Leaving element <target>
gdbarch_find_by_info: info.bfd_arch_info i386:x86-64
gdbarch_find_by_info: info.byte_order 1 (little)
gdbarch_find_by_info: info.osabi 5 (GNU/Linux)
gdbarch_find_by_info: info.abfd 0x556fd3f9f2d0
gdbarch_find_by_info: info.tdep_info 0x0
gdbarch_find_by_info: New architecture 0x556fd40dfed0 (i386:x86-64) selected
gdbarch_dump: GDB_NM_FILE = config/nm-linux.h
gdbarch_dump: addr_bit = 64
gdbarch_dump: addr_bits_remove = <0x556fd321b910>
gdbarch_dump: gdbarch_address_class_name_to_type_flags_p() = 0
gdbarch_dump: address_class_name_to_type_flags = <0x0>
gdbarch_dump: gdbarch_address_class_type_flags_p() = 0
gdbarch_dump: address_class_type_flags = <0x0>
gdbarch_dump: gdbarch_address_class_type_flags_to_name_p() = 0
gdbarch_dump: address_class_type_flags_to_name = <0x0>
[... 307 lines cut ...]
Sending packet: $m40003d01dc4b,1#e3...Packet received: 48
remote:target_xfer_partial (1, (null), 0x7fff771ef8a0, 0x0, 0x40003d01dc4b, 1) = 1, 1, bytes =
 48
Sending packet: $Z0,40003d01dc4b,1#2c...Packet received: 
Packet Z0 (software-breakpoint) is NOT supported
Sending packet: $m40003d01dc4b,1#e3...Packet received: 48
remote:target_xfer_partial (1, (null), 0x7fff771ef8c0, 0x0, 0x40003d01dc4b, 1) = 1, 1, bytes =
 48
Sending packet: $X40003d01dc4b,0:#07...Packet received: 
binary downloading NOT supported by target
Sending packet: $M40003d01dc4b,1:cc#c3...Packet received: OK
remote:target_xfer_partial (2, (null), 0x0, 0x556fd3689d40, 0x40003d01dc4b, 1) = 1, 1, bytes =
 cc
Sending packet: $vCont;c:p1.-1#0f...Packet received: T05thread:p01.01;hwbreak:;
Sending packet: $g#67...Packet received: 0000000000000000000000000000000030bc493d00400000160000000000000049ce403d00400000308d423e00400000f049393e004000000048393e0040000000000000000000000000000000000000a83a393e00400000a6000000000000000100000000000000010000000000000004000000000000005403203f000000004cdc013d004000000600000008000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000007f0300000000000000000000[608 bytes omitted]
target_fetch_registers (rip) = 4cdc013d00400000 0x40003d01dc4c 70369767709772
Sending packet: $P10=4bdc013d00400000#c7...Packet received: OK
Packet P (set-register) is supported
target_store_registers (rip) = 4bdc013d00400000 0x40003d01dc4b 70369767709771
Sending packet: $P39=ffffffffffffffff#59...Packet received: 
Sending packet: $M40003d01dc4b,1:48#69...Packet received: OK
remote:target_xfer_partial (2, (null), 0x0, 0x556fd421fdc4, 0x40003d01dc4b, 1) = 1, 1, bytes = 48
Protocol error: P (set-register) conflicting enabled responses.
(gdb) 
```

gz, 2021-09-26:
Now that I look at it, this might be due to the osabi in the elf file...

```
gdbarch_find_by_info: info.osabi 5 (GNU/Linux)
gdbarch_find_by_info: info.abfd 0x556fd3f9f2d0
gdbarch_find_by_info: info.tdep_info 0x0
gdbarch_find_by_info: New architecture 0x556fd40dfed0 (i386:x86-64) selected
gdbarch_dump: GDB_NM_FILE = config/nm-linux.h
```

daniel5151, 2021-09-26:
Ahh, interesting.

That would line up with your earlier guess that this might be related to https://github.com/bminor/binutils-gdb/blob/master/gdb/features/i386/64bit-linux.xml#L10, given that the value being written in a 64-bit `0xffffffffffffffff`.

What happens if you run `set osabi none` prior to stepping?

daniel5151, 2021-09-27:
Any update on this?

gz, 2021-09-27:
Yes, I tried the `set osabi none` but if I set that I receive the infamous `g packet too long` response in gdb and things stopped working immediately after. So I gave up on that and tried to find out how to tell the rust linker not to define my kernel ELF as GNU/Linux -- found out that it's not even doing that (it sets it to SystemV Unix) -- tried to figure out how to change that and ultimately gave up :).

As a simple hack I'll probably set the OS ABI in ELF to none with the hex editor just to see what would happen, but got distracted so haven't tried yet.

daniel5151, 2021-09-27:
Hmmm, very weird indeed.

Let me know if you keep looking into this, as while there is the disclaimer in `gdbstub_arch` that implementations aren't guaranteed to be correct, it would be best if they _were_ correct :)

* * *

Also, in the future, could you wrap log output in a `<details>` tag? I've gone ahead and edited your past comments to do that, as it does make it easier to skim through the issue discussion.

i.e: the markdown should look like:

````markdown
<details>
<summary>gdb log</summary>

```
// log text
```

</defails>
````

gz, 2021-09-27:
Yes I'll definitely follow up this week on this issue. I'd def. want to use the `read/write_register` as opposed to `read/write_registers` for efficiency, and currently my gdb stops doing working with `read/write_register` enabled after it tries to set the reg 57 and gets a response that it can't do that.

One step before that is that I need to get step/next to work properly. Seems that gdb will want over-write some instructions for that in .text, which in my case is mapped as read-only so not an option. But interestingly it *looks like* if you tell gdb that it can't write there it falls back to single-stepping and doesn't just set hardware breakpoints instead (that all be me doing it wrong though so I'll have to look into it some more first).

daniel5151, 2021-09-27:
It should be noted that the GDB `step` command doesn't map onto the `step` packet in the GDB RSP (or at least, it doesn't _have_ to).

The GDB `step`/`next` commands are source-code level directives. i.e: single step a single _line_ of code. As such, at the protocol level, GDB will actually set a breakpoint, send a `continue` packet, and then unset the breakpoint once it's hit.

On the other hand, the GDB `stepi` command will actually single step over _instructions_, and uses the `step` packet in the GDB RSP (instead of `continue` + breakpoints).

As for when GDB tries to use hardware vs. software breakpoints... that is something I'm a bit fuzzy on. My hands-on experience integrating `gdbstub` into projects has primarily been with _emulators_, where breakpoints are something handled "unintrusively" via the emulator's interpreter infrastructure. As such, the challenges you're running into are fairly "new ground" for `gdbstub`, and as AFAIK, only you and xobs (in #56) have done this sort of thing 😅

In other words: your exploration and efforts are much appreciated, as you're providing a very useful practical validation of `gdbstub`'s current bare-metal debugging capabilities!

gz, 2021-09-28:
Thanks that's helpful! 

It was surprising to learn that it would still go off and insert int3 instructions in random locations (without software breakpoints enabled). But, implementing SwBreak by using hardware breakpoints is a good idea :)!

daniel5151, 2021-09-28:
Indeed, it is very surprising! This is actually something xobs came across in https://github.com/daniel5151/gdbstub/issues/56#issuecomment-868205098, and is definitely something I'll need to clarify in the documentation. It's quite the footgun!

* * *

Note that you _may_ be able to force the GDB client to automatically use hardware breakpoints by implementing the `SwBreakpoints` IDET to always return `Ok(false)` (or possibly returning an `Err` or some kind). This _may_ result in the GDB client falling back to trying to set a hwbreak.

Strong emphasis on the _may_, as this behavior is very much up to the whims of the GDB client implementation, and nailing down the precise example will require some experimentation and/or reading through the GDB client source.

gz, 2021-09-28:
Ah good point, I'll try that and report back :)

daniel5151, 2021-11-04:
I know this issue got a bit sidetracked, so lets refocus and understand the heart of the issue:

Were you able to figure out why your particular target / arch / binary was requesting that extra register?

gz, 2021-11-04:
Not yet -- so far I just disabled the single register read/write. But I'll focus on this again now!

gz, 2021-11-05:
Hm, I wish I had documented the exact steps that led to this failure. I re-enabled single register read/write but can't trigger this error anymore. (I'm not getting any write register requests out-of-the-blue anymore (good), except if I trigger them myself using something like `set $reg = 0x99` in gdb). So I'll close it.

closed 2021-11-05 by gz