daniel5151/gdbstub #89: PacketUnexpected with ThreadStopReason::GdbCtrlCInterrupt

gz, 2021-10-01:
I tried to support Ctrl+C (in GDB) by waiting for the serial line interrupt, and if it arrives providing a `ThreadStopReason::GdbCtrlCInterrupt` to the state machine. However, this gives me:

```
17017938078 [ERROR] - gdbstub::gdbstub_impl::state_machine: Unexpected interrupt packet!
17029278514 [ERROR] - nrk::arch::gdb: gdbstub internal error PacketUnexpected
```

Which seems to be going [here](https://github.com/daniel5151/gdbstub/blob/4e46b723d757ef3577aed69d2394c44da063f8cb/src/gdbstub_impl/mod.rs#L435). The comment is strange because I'm doing a Ctrl+C while my code is [spinning in an infinite loop](https://github.com/vmware-labs/node-replicated-kernel/pull/67/commits/c943ca493d2e6832ccaac90ab9808606de46ff13#diff-c8df6312627090b46e3a1bd399bf9925fd36b9e307e050ba7a67555c464cc21fR512). Anyways I figured I'd ask also because PacketUnexpected is treated as ["always a bug"](https://github.com/daniel5151/gdbstub/blob/5c702d063baaf78e4ebed7842cc8b96195d119a8/src/gdbstub_impl/error.rs#L22)...

daniel5151, 2021-10-01:
Yeah... that's not good.

Could you post some more context on how you arrived at this situation? i.e: the debug logs from `gdbstub` and the GDB client?

gz, 2021-10-01:
<details>
<summary>gdb log</summary>

```log
 839434646 [INFO ] - nrk::arch::gdb: Waiting for a GDB connection (I/O port 0x2f8
)...                                                                             
 847761916 [INFO ] - nrk::arch::gdb: Use `target remote localhost:1234` in gdb to
 connect.                                                                        
142583929118 [INFO ] - nrk::arch::gdb: Debugger connected.                       
142627577238 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("vMustReplyEmpt
y")                                                                              
142770680278 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("qTStatus")    
142877358354 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("qC")          
142885060668 [TRACE] - nrk::arch::gdb: read_registers X86_64CoreRegs { regs: [00,
 00, 40003D4939F0, 16, 40003D1FD7A2, 40003E426CE8, 40003E426FF0, 40003E355000, 00
, 00, 40003E3540A8, 591, 01, 01, 04, 3F2005D4], eflags: 06, rip: 40003D0024EC, se
gments: X86SegmentRegs { cs: 08, ss: 10, ds: 00, es: 00, fs: 00, gs: 00 }, st: [[
00, 00, 00, 00, 00, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
, [00, 00, 00, 00, 00, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 
00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00, 0
0, 00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00
, 00, 00]], fpu: X87FpuInternalRegs { fctrl: 37F, fstat: 00, ftag: 00, fiseg: 00,
 fioff: 00, foseg: 00, fooff: 00, fop: 00 }, xmm: [40003CA1C000000040003CA1C000, 
20000000000000000000000000, 40003D422620, C0C0, 8080, 10000000000000001, 00, 00, 
00, 00, 00, 00, 00, 00, 00, 00], mxcsr: 1FA0 }                                   
142925585414 [TRACE] - nrk::arch::gdb: read_addr 0x57cf40                        
142932373264 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("vFile:setfs:0"
)                                                                                
143031828638 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("vFile:open:6a7
573742070726f62696e67,0,1c0")                                                    
143054849962 [TRACE] - nrk::arch::gdb: read_addr 0x40003d0024ec                  
143062693116 [TRACE] - nrk::arch::gdb: read_addr 0x40003d0024ec                  
143084145190 [TRACE] - nrk::arch::gdb: read_addr 0x40003e426d98                  
143187032914 [TRACE] - nrk::arch::gdb: read_addr 0x40003e426da0                  
143193488566 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("qSymbol::")   
151065824906 [TRACE] - nrk::arch::gdb: resume_with =  Continue                   
151068806666 [TRACE] - nrk::arch::gdb: Resume execution.                         
151072033922 [INFO ] - nrk: watchpoint_trigger is 0                              
151074625712 [INFO ] - nrk: watchpoint_trigger is 3735928559                     
151077452858 [INFO ] - nrk: step                                                 
151079547652 [INFO ] - nrk: step                                                 
157326735746 [DEBUG] - gdbstub::gdbstub_impl: <-- interrupt packet               
157329869902 [ERROR] - gdbstub::gdbstub_impl::state_machine: Unexpected interrupt packet!                                                                         
157333321616 [ERROR] - nrk::arch::gdb: gdbstub internal error PacketUnexpected  
```
</details>

<details>
<summary>gdb log</summary>

```log
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ../target/x86_64-uefi/debug/esp/kernel...
-> exec->log_command (...)
<- exec->log_command (target remote localhost:1234)
-> remote->open (...)
target_close ()
[... 324 lines cut ...]
remote:target_xfer_partial (1, (null), 0x7ffd7e77c6f7, 0x0, 0x40003d0024ec, 1) = 1, 1, bytes = e8
Sending packet: $m40003d0024ec,1#b3...Packet received: e8
remote:target_xfer_partial (1, (null), 0x7ffd7e77c6f7, 0x0, 0x40003d0024ec, 1) = 1, 1, bytes = e8
nrk::arch::_start (Sending packet: $m40003e426d98,8#9e...Packet received: 0050123e00400000
remote:target_xfer_partial (1, (null), 0x55c0b7ea7830, 0x0, 0x40003e426d98, 8) = 1, 8, bytes =
 00 50 12 3e 00 40 00 00
argc=70369785565184, Sending packet: $m40003e426da0,8#be...Packet received: 0900000000000000
remote:target_xfer_partial (1, (null), 0x55c0b7b82a10, 0x0, 0x40003e426da0, 8) = 1, 8, bytes =
 09 00 00 00 00 00 00 00
_argv=0x9) at kernel/src/arch/x86_64/mod.rs:927
927	    let _r = xmain();
Sending packet: $qSymbol::#5b...Packet received: 
Packet qSymbol (symbol-lookup) is NOT supported
<- remote->open (localhost:1234, 0)
(gdb) continue
Continuing.
Sending packet: $vCont?#49...Packet received: vCont;c;C;s;S
Packet vCont (verbose-resume) is supported
Sending packet: $vCont;c:p1.-1#0f...^Cremote_pass_ctrlc called
remote_interrupt called
Packet received: S05
Sending packet: $g#67...target_close ()
Remote connection closed
```
</details>

daniel5151, 2021-10-01:
Ahh, sorry - could you re-upload the gdbstub trace log with the `trace-pkt` feature enabled?

daniel5151, 2021-10-01:
Also, could you link to the specific revision of your `gdbstub` implementation that exhibits this behavior?

gz, 2021-10-01:
I'm on the `refactor/blocking-event-loop` currently, will check if this also happens with latest dev/0.6 in a second

<details>
<summary>gdbstub with trace-pkt</summary>

```log
1066765958 [INFO ] - nrk::arch::gdb: Use `target remote localhost:1234` in gdb to connect.                                                                                                                                   
81092034144 [INFO ] - nrk::arch::gdb: Debugger connected.                                                                                                                                                                    
81096823916 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                                                                                  
81124022406 [TRACE] - gdbstub::protocol::recv_packet: <-- $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;xmlRegisters=i386#6a    
81145838950 [TRACE] - gdbstub::protocol::response_writer: --> $PacketSize=1000;vContSupported+;multiprocess+;QStartNoAckMode+;swbreak+;hwbreak+;qXfer:features:read+#47                                                      
81153880874 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                                                                                  
81159061360 [TRACE] - gdbstub::protocol::recv_packet: <-- $vMustReplyEmpty#3a                                                                                                                                                
81162802302 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("vMustReplyEmpty")                                                                                                                                          
81166160342 [TRACE] - gdbstub::protocol::response_writer: --> $#00                                                                                                                                                           
81169668558 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                                                                                  
81174859122 [TRACE] - gdbstub::protocol::recv_packet: <-- $QStartNoAckMode#b0                                                                                                                                                
81180146078 [TRACE] - gdbstub::protocol::response_writer: --> $OK#9a                                                                                                                                                         
81185443954 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                                                                                  
81190546912 [TRACE] - gdbstub::protocol::recv_packet: <-- $Hgp0.0#ad                                                                                                                                                         
81195421004 [TRACE] - gdbstub::protocol::response_writer: --> $OK#9a                                                                                                                                                         
81294244030 [TRACE] - gdbstub::protocol::recv_packet: <-- $qXfer:features:read:target.xml:0,ffb#79                                                                                                                           
81304113196 [TRACE] - gdbstub::protocol::response_writer: --> $m<target version="1.0"><architecture>i386:x86-64</architecture><feature name="org.gnu.gdb.i386.sse"></feature></target>#c7                                    
81317905154 [TRACE] - gdbstub::protocol::recv_packet: <-- $qXfer:features:read:target.xml:76,ffb#b6                                                                                                                          
81323953014 [TRACE] - gdbstub::protocol::response_writer: --> $l#6c                                                                                                                                                          
81343168780 [TRACE] - gdbstub::protocol::recv_packet: <-- $qTStatus#49                                                                                                                                                       
81347922584 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("qTStatus")                                                                                                                                                 
81352925762 [TRACE] - gdbstub::protocol::response_writer: --> $#00                                                                                                                                                           
81447439352 [TRACE] - gdbstub::protocol::recv_packet: <-- $?#3f                                                                                                                                                              
81450929164 [TRACE] - gdbstub::protocol::response_writer: --> $S05#b8                                                                                                                                                    [... cut ...]
94954960864 [INFO ] - nrk: step                                                  
94958019902 [INFO ] - nrk: step                                                  
102264921268 [TRACE] - gdbstub::protocol::response_writer: --> $S05#b8           
102268107020 [TRACE] - gdbstub::protocol::recv_packet: <--                       
102271156160 [DEBUG] - gdbstub::gdbstub_impl: <-- interrupt packet               
102273952962 [ERROR] - gdbstub::gdbstub_impl::state_machine: Unexpected interrupt packet!
```

</details>

gz, 2021-10-01:
Same issue with `dev/0.6` (I guess this was at the same revision anyways). Maybe I'm also just doing something stupid. My approach is to wait for the serial line receive interrupt and if I get that I pass in the `ThreadStopReason::GdbCtrlCInterrupt` to the state machine. Is this what I'm supposed to do?

daniel5151, 2021-10-01:
Before I look into it this more (likely sometime this weekend), do you know if this was something that started happening after I merged #88, or was this happening before as well?

Theoretically, this assert is there because the GDB client should never send an interrupt packet while the GDB stub is still in the "paused" state (i.e: the `Pump` state, where it is waiting for more data while the target isn't running). The fact that this is happening means that there's either an issue with how the state machine API is written (i.e: there's a missing state transition), or there's something spooky going on in your implementation (which I would appreciate a commit-hash'd link to)

gz, 2021-10-01:
Here is the start of my state machine:

https://github.com/vmware-labs/node-replicated-kernel/blob/c943ca493d2e6832ccaac90ab9808606de46ff13/kernel/src/arch/x86_64/gdb.rs#L101

pretty much all the gdb logic stuff in that file

gz, 2021-10-01:
I didn't have `Ctrl+C` working before #88 but I can easily revert the changes for that branch and try it, let me do that.

daniel5151, 2021-10-02:
Alright, I've skimmed through your implementation, and I think I understand the issue in your implementation.

Before going into the specific issue at hand, lets back up for a second and discuss how the overall state machine flow goes: 

```
                        (1) pump(u8)
                 ┌─────────────────────────┐
                 │                         │
  start    ┌─────┴──┐                 ┌────▼─────────────────┐
 ──────────►  Pump  │                 │  DeferredStopReason  │
           └─────▲──┘                 └────┬───┬──────────▲──┘
                 │                         │   │          │
                 └─────────────────────────┘   └──────────┘
                   deferred_stop_reason(..)    (2) pump(u8)
```

The `Pump` state represents the state where the Target is entirely stopped, and `gdbstub` is waiting for data (e.g: "read these registers", "write to this memory address", etc...).

Once the `gdbstub` receives a resume packet, it transitions into the `DeferredStopReason` state, whereupon the Target is running _concurrently_ with `gdbstub`.

In this state, there are _two_ events that need to be handled (hence the two methods on`GdbStubStateMachineInner<DeferredStopReason>`: the case where the Target reports a stop reason (e.g: a breakpoint is hit), **and** the case where the GDB client sends more data down the line.

The first case is easy to understand: call `deferred_stop_reason` with the appropriate stop reason, and then the stub will transition back into `Pump`, awaiting more instructions.

The second case seems a bit weird... why would the GDB client send data while the target is running?

Well... as it turns out, Ctrl-C interrupts are signaled by sending special "interrupt" packets down the wire! Specifically, an interrupt can be signaled either by sending a single `0x3` byte down the wire, or in some cases, by sending a properly framed `vCtrlC` packet.

See https://sourceware.org/gdb/onlinedocs/gdb/Interrupts.html#Interrupts for details.

* * *

In your implementation, you are not handling the "data is sent down the wire while the target is running" case _at all_. Instead, your current implementation seems to do the following:

1. `KCoreStopReason::ConnectionInterrupt => Some(ThreadStopReason::GdbCtrlCInterrupt)` - i.e: if you sense new data on the wire, _without even checking what it is_, report a stop reason.
2. `deferred_stop_reason` dutifully reports the stop reason back to the client, and transitions back into the `Pump` state to await new data
3. Upon entering the `Pump` state, it finally reads the data that was sent down the wire, only to find a interrupt packet that _should_ have been handled by the `pump` flow in the `DeferredStopReason` state! An assertion fires, and everything dies.

i.e: you should only call `deferred_stop_reason` after the `.pump()` method in the `DeferredStopReason` returns a `Event::CtrlCInterrupt`.

A good starting point for how to structure your event loop would be to look at the guts of the new `run_blocking` method, specifically, the contents of the `GdbStubStateMachine::DeferredStopReason` match arm:

https://github.com/daniel5151/gdbstub/blob/4e46b72/src/gdbstub_impl/mod.rs#L199

* * *

After writing this explanation, I realize that the fact that you ran into this issue is indicative that this new API still needs some work + polish. 

I suspect that the API itself could be improved (possibly by adding some more fine-grained states), and it goes without saying that the biggest missing piece right now is good documentation/examples...

Ideally, I want to craft an API that's impossible to misuse - leveraging as many Rust type-system shenanigans to have the compiler enforce all branches are being accounted for. I'll have to think about how I can tweak the current iteration of the API to achieve that...

daniel5151, 2021-10-02:
Ah, in addition, I really should remove the `ThreadStopReason::GdbCtrlCInterrupt` entirely. It's literally just a "short hand" for writing `ThreadStopReason::DoneStep`, but I suspect that its naming will lead to lots of confusion...

I'll go ahead and do immediately.

gz, 2021-10-02:
Ah I see, yes I figured the chances are high it's just me doing something wrong here. This detailed explanation helps a lot. If you'd like me to I could try to put this into the docs (only if you want to do it yourself of course).

daniel5151, 2021-10-02:
I appreciate the offer, but I think this is something I should tackle myself - especially since there's a non-zero change I rework the API a bit, which would invalidate any docs.

And of course, when you get the chance, please let me know if my explanation + suggested fix was correct!

gz, 2021-10-06:
Great, so thanks a lot for your help, I [rewrote my state-machine a little](https://github.com/vmware-labs/node-replicated-kernel/pull/67/commits/b33d52c3db775f0d21d03de33a1be85854b56b88) to reflect the fact that it should pump() in DeferredStopReason when I have data available. I hope this accurately captures what you wanted me to do.

With this change, `Ctrl+C` does work fine together with `continue` (tested with an infinite loop):

```
nrk::arch::_start (argc=70369785565184, _argv=0x9)
    at kernel/src/arch/x86_64/mod.rs:927
927         let _r = xmain();
(gdb) continue
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
nrk::xmain () at kernel/src/integration_main.rs:512
512         while cond {}
(gdb) continue
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
nrk::xmain () at kernel/src/integration_main.rs:512
512         while cond {}
(gdb) continue
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
nrk::xmain () at kernel/src/integration_main.rs:512
512         while cond {}
(gdb) next
^CRemote communication error.  Target disconnected.: Connection reset by peer.
```

So while `continue` works fine, doing a `next` or `step` and then hitting Ctrl+C again still gives me `gdbstub::gdbstub_impl::state_machine: Unexpected interrupt packet!` (and the connection reset above as a cascading error). 

I think I understand what's going on: when I do `next`, the program will start to single-step through the loop so it will execute the state-machine logic many times. What's weird then is that the state-machine does end up in the Pump state (as indicated by the [`pumping...` log message](https://github.com/vmware-labs/node-replicated-kernel/pull/67/commits/b33d52c3db775f0d21d03de33a1be85854b56b88#diff-eeeb3cc09458fbf553ad4bf27bb10f944b67e5181c122af65b0fdfa42391a719R106) below. Am I correct that this shouldn't happen since GDB is just single-stepping and I don't actually have a prompt (in gdb)?

e.g. the way the log looks it seems for one state-machine execution after taking a step it ends up many times in Pump state without me ever having a GDB prompt, so unsurprisingly it gets an unexpected interrupt packet when I hit Ctrl+c during this time:

```
[ .. beginning of one gdb stm logic when we got interrupted from after taking a  single-step ..]
26533530728 [INFO ] - nrk::arch::gdb: event_loop reason DebugInterrupt               
[.. we determine that we stopped because we're stepping ..]
26537096730 [TRACE] - nrk::arch::gdb: stop reason is DoneStep                        
26540296784 [INFO ] - nrk::arch::gdb: stop_reason Some(DoneStep) 
[ .. the stm was in deferredstopreason state, this is good too ..]                    
26543815968 [INFO ] - nrk::arch::gdb: deferred_stop_reason                           
26547158636 [TRACE] - gdbstub::protocol::response_writer: --> $S05#b8
[ .. transitions into pump state here, ok but I don't have a gdb prompt yet ..]                
26550949190 [INFO ] - nrk::arch::gdb: pumping...                                     
26554247238 [INFO ] - nrk::arch::gdb: pumping...                                     
26557388140 [INFO ] - nrk::arch::gdb: pumping...                                     
26560478940 [INFO ] - nrk::arch::gdb: pumping...                                     
26563497220 [INFO ] - nrk::arch::gdb: pumping...                                     
26566565346 [TRACE] - gdbstub::protocol::recv_packet: <-- $g#67  
[  .. does some stuff this is all fine ..]
26570024202 [TRACE] - nrk::arch::gdb: read_registers X86_64CoreRegs { regs: [40003D4B
FC4F, 00, 40003D41D600, 01, 00, 40003D4BFC4F, 40003E354FF0, 40003E354D70, 00, 00, 400
03E3544F8, 8C, 01, 01, 04, 3F2005D4], eflags: 302, rip: 40003CFB5CB4, segments: X86Se
gmentRegs { cs: 08, ss: 10, ds: 00, es: 00, fs: 00, gs: 00 }, st: [[00, 00, 00, 00, 0
0, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00
, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00,
 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 
00, 00, 00, 00, 00], [00, 00, 00, 00, 00, 00, 00, 00, 00, 00]], fpu: X87FpuInternalRe
gs { fctrl: 37F, fstat: 00, ftag: 00, fiseg: 00, fioff: 00, foseg: 00, fooff: 00, fop
: 00 }, xmm: [1000040003D49BC88, 20000000000000000000000000, 40003D41D860, C0C0, 8080
, 10000000000000001, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00], mxcsr: 1FA0 }          
26609124980 [TRACE] - gdbstub::protocol::response_writer: --> $4ffc4b3d00400000000000
000000000000d6413d00400000010000000000000000000000000000004ffc4b3d00400000f04f353e004
00000704d353e0040000000000000000000000000000000000000f844353e004000008c00000000000000
010000000000000001000000000000000400000000000000d405203f00000000b45cfb3c0040000002030
0000800000010000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[... 51 lines cut ...]
: 00 }, xmm: [1000040003D49BC88, 20000000000000000000000000, 40003D41D860, C0C0, 8080
, 10000000000000001, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00], mxcsr: 1FA0 }          
27098559170 [TRACE] - gdbstub::protocol::response_writer: --> $4ffc4b3d00400000000000
000000000000d6413d00400000010000000000000000000000000000004ffc4b3d00400000f04f353e004
00000704d353e0040000000000000000000000000000000000000f844353e004000008c00000000000000
[... cut ...]
[.. here I hit Ctrl+C in GDB (I didn't have a prompt) .. ]                                     
27139247866 [TRACE] - gdbstub::protocol::recv_packet: <--                            
27142773072 [DEBUG] - gdbstub::gdbstub_impl: <-- interrupt packet
[ .. gdbstub (rightfully?) complains because it was in pump state? ..]                    
27146204800 [ERROR] - gdbstub::gdbstub_impl::state_machine: Unexpected interrupt pack
et!
27150376962 [ERROR] - nrk::arch::gdb: gdbstub internal error PacketUnexpected      
```

daniel5151, 2021-10-06:
Hmm, I'm not entirely sure what's going on here just yet, but here are a few things you can try:

1. Blindly copy the code from https://github.com/daniel5151/gdbstub/blob/dev/0.6/src/gdbstub_impl/mod.rs#L236-L252, explicitly calling `deferred_stop_reason` from the `Event::CtrlCInterrupt` branch.
2. Maybe your single-step implementation has a bug in it? If so, you can try out the changes at #92, keep single-stepping disabled, and see if that makes the issue go away.
    - While I would appreciate you to test out that code, if you're lazy, you can also just temporarily fork `gdbstub` again, and modify https://github.com/daniel5151/gdbstub/blob/dev/0.6/src/gdbstub_impl/ext/base.rs#L301 to return just `vCont;c;C`

gz, 2021-10-06:
Did some tests:

Re 1: [Changed it](https://github.com/vmware-labs/node-replicated-kernel/pull/67/commits/2ecb11802eceec72c7d664ef913630fd05cf460b) but didn't make a difference (still seems like it gets a ctrl+c packet when not expecting it)

Re 2: [Using the new feature branch (`feature/optional-single-step`)](https://github.com/vmware-labs/node-replicated-kernel/pull/67/commits/3e9253042d6246613d0f03646c38810e3a6d15be): Didn't resolve the issue, I noticed that disabling stepping breaks my `next` and `step` commands a bit more than it is alrready now :) (meaning I can't even get to point where I would issue a ctrl+c to stop my infinite loop).

With stepping is enabled, the `step`, `next` commands works fine (like before I switched to the branch) when I'm not in a loop (but it will end up with an unexpected packet error when I'm sending a ctrl+c after a `next` in an infinite loop):

```
nrk::arch::_start (argc=70369785565184, _argv=0x9)                                    
    at kernel/src/arch/x86_64/mod.rs:927                                              
927         let _r = xmain();                                                         
(gdb) step                                                                            
nrk::xmain () at kernel/src/integration_main.rs:499                                   
499         arch::irq::ioapic_establish_route(0x0, 0x0);
(gdb) step
nrk::arch::irq::ioapic_establish_route (_gsi=0, _core=0)
    at kernel/src/arch/x86_64/irq.rs:810
810         for io_apic in atopology::MACHINE_TOPOLOGY.io_apics() {
(gdb) step
<atopology::MACHINE_TOPOLOGY as core::ops::deref::Deref>::deref (self=0x40003d409b38)
    at /home/gz/.cargo/registry/src/github.com-1ecc6299db9ec823/lazy_static-1.4.0/src/lib.rs:144
144                     __stability()
(gdb) step
<atopology::MACHINE_TOPOLOGY as core::ops::deref::Deref>::deref::__stability ()
    at /home/gz/.cargo/registry/src/github.com-1ecc6299db9ec823/lazy_static-1.4.0/src/lib.rs:144
144                     __stability()
(gdb) step
lazy_static::lazy::Lazy<T>::get (
    self=0x40003d4bfb08 <<atopology::MACHINE_TOPOLOGY as core::ops::deref::Deref>::deref::__stability::LAZY>, builder=<optimized out>)
    at /home/gz/.cargo/registry/src/github.com-1ecc6299db9ec823/lazy_static-1.4.0/src/core_lazy.rs:21
21              self.0.call_once(builder)
(gdb) step
```

If I disable stepping (`fn support_single_step` returns None), I'm getting an unexpected packet error immediately after sending a `next` in gdb (similar problem to the one above but this time it's not a ctrl+c packet but rather a vCont package):
```
10122983318 [INFO ] - nrk::arch::gdb: pumping...                                      
10126255784 [INFO ] - nrk::arch::gdb: pumping...                                      
10129504230 [INFO ] - nrk::arch::gdb: pumping...                                      
10132724428 [TRACE] - gdbstub::protocol::recv_packet: <-- $vCont;s:p1.1;c:p1.-1#f7    
10136879558 [ERROR] - nrk::arch::gdb: gdbstub internal error PacketUnexpected   
```

gz, 2021-10-07:
And for having single-stepping disabled. I haven't shared a complete log for that failure yet so here is one:

<details>
<summary>gdbstub trace log</summary>

```
1101136270 [INFO ] - nrk::arch::gdb: Use `target remote localhost:1234` in gdb to connect.                                                                     
10680332932 [INFO ] - nrk::arch::gdb: Debugger connected.                                                                                                      
10685063254 [TRACE] - nrk::arch::gdb: event_loop reason DebugInterrupt                                                                                         
10688828938 [TRACE] - nrk::arch::gdb: stop_reason None                                                                                                         
10691941324 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                    
10719219804 [TRACE] - gdbstub::protocol::recv_packet: <-- $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vCont
Supported+;QThreadEvents+;no-resumed+;xmlRegisters=i386#6a                                                                                                     
10740276684 [TRACE] - gdbstub::protocol::response_writer: --> $PacketSize=1000;vContSupported+;multiprocess+;QStartNoAckMode+;swbreak+;hwbreak+;qXfer:features:
read+#47                                                                                                                                                       
10749386038 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                    
10755553658 [TRACE] - gdbstub::protocol::recv_packet: <-- $vMustReplyEmpty#3a                                                                                  
10760667638 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("vMustReplyEmpty")                                                                            
10765739762 [TRACE] - gdbstub::protocol::response_writer: --> $#00                                                                                             
10769783520 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                    
10775149526 [TRACE] - gdbstub::protocol::recv_packet: <-- $QStartNoAckMode#b0                                                                                  
10779194636 [TRACE] - gdbstub::protocol::response_writer: --> $OK#9a                                                                                           
10783279600 [TRACE] - gdbstub::protocol::recv_packet: <-- +                                                                                                    
10787410004 [TRACE] - gdbstub::protocol::recv_packet: <-- $Hgp0.0#ad                                                                                           
10792295672 [TRACE] - gdbstub::protocol::response_writer: --> $OK#9a                                                                                           
10893187020 [TRACE] - gdbstub::protocol::recv_packet: <-- $qXfer:features:read:target.xml:0,ffb#79                                                             
10906524372 [TRACE] - gdbstub::protocol::response_writer: --> $m<target version="1.0"><architecture>i386:x86-64</architecture><feature name="org.gnu.gdb.i386.s
se"></feature></target>#c7                                                                                                                                     
10923025926 [TRACE] - gdbstub::protocol::recv_packet: <-- $qXfer:features:read:target.xml:76,ffb#b6                                                            
10927686182 [TRACE] - gdbstub::protocol::response_writer: --> $l#6c                                                                                            
10945510934 [TRACE] - gdbstub::protocol::recv_packet: <-- $qTStatus#49                                                                                         
10950528166 [INFO ] - gdbstub::gdbstub_impl: Unknown command: Ok("qTStatus")                                                                                   
10955737284 [TRACE] - gdbstub::protocol::response_writer: --> $#00                                                                                             
11054542430 [TRACE] - gdbstub::protocol::recv_packet: <-- $?#3f                                                                                                
11058036250 [TRACE] - gdbstub::protocol::response_writer: --> $S05#b8                                                                                          
11063477958 [TRACE] - gdbstub::protocol::recv_packet: <-- $qfThreadInfo#bb                                                                                     
11067548232 [TRACE] - gdbstub::protocol::response_writer: --> $mp01.01#cd                                                                                      
11073135704 [TRACE] - gdbstub::protocol::recv_packet: <-- $qsThreadInfo#c8                                                                                     
11077081434 [TRACE] - gdbstub::protocol::response_writer: --> $l#6c                                                                                            
[... cut ...]
remote:target_xfer_partial (1, (null), 0x55ea2712ff20, 0x0, 0x40003e426da0, 8) = 1, 8, bytes =
 09 00 00 00 00 00 00 00
_argv=0x9)
    at kernel/src/arch/x86_64/mod.rs:927
927         let _r = xmain();
Sending packet: $qSymbol::#5b...Packet received: 
Packet qSymbol (symbol-lookup) is NOT supported
<- remote->open (localhost:1234, 0)
(gdb) next
Sending packet: $vCont?#49...Packet received: vCont;c;C
Packet vCont (verbose-resume) is supported
Sending packet: $vCont;s:p1.1;c:p1.-1#f7...target_close ()
Remote connection closed
(gdb) 
```

</details>

gz, 2021-10-07:
Yes this is using the `feature/optional-single-step` branch, and looks like you're guess is correct:

```
5660085612 [TRACE] - gdbstub::protocol::response_writer: --> $vCont;c;C#265674929254 
[TRACE] - gdbstub::protocol::recv_packet: <-- $vCont;s:p1.1;c:p1.-1#f7
5685461636 [INFO ] - gdbstub::gdbstub_impl::ext::base: we are at base.rs:510
5695651002 [ERROR] - nrk::arch::gdb: gdbstub internal error PacketUnexpected
```

gz, 2021-10-07:
```
gdb --version
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
```

gz, 2021-10-07:
> Well, aside from the possibly weird interaction of returning a HwBreak stop reason from a breakpoint set via z0/Z0, good to see that it seems to be working as expected...

Right, I noticed that too that I should probably respond with SwBreak. This is a problem with me forwarding this just to the HwBreak logic, will fix it. In theory it shouldn't matter for the vCont issue as this won't even set. But I'll fix it now and report back

daniel5151, 2021-10-08:
Right. Well shoot, I'm stumped.

Unless there's something incredibly obvious I'm missing, it seems that GDB is ignoring the fact that the stub doesn't support single-stepping, and is sending it single-step commands regardless.

I'll try to find some time this weekend to whip up some kind of "dummy" `gdbstub` implementation that uses the same `Arch` as you're currently using, and see if I can repro the behavior locally. Maybe this is some weird x86 specific issue, whereby GDB will unconditionally assume single-stepping is supported at the architecture level? If that's the case, then I'll have to rethink some of the changes I'm making in #92, in order to avoid try and catch what might be a _absolutely terrible_ footgun.

* * *

In the meantime, I'll suggest that we shelve this digression into why optional single-stepping isn't working, and re-focus on the actual issue at hand: getting Ctrl-C interrupts working.

Go ahead and re-enable support for single stepping support, and then see if you can get your state machine implementation working correctly _without_ handling Ctrl-C interrupts. That should give us a baseline level of "working code" to then re-introduce Ctrl-C functionality into.

daniel5151, 2021-10-09:
Well I'll be damned. 

I hacked together a dummy `gdbstub` implementation backed by the x86 arch (i.e: memory is entirely filled with NOPs, registers are set to reasonable dummy values), and lo and behold: invoking `stepi` will unconditionally send `s` to the stub!

I'm now digging through the GDB client code to figure out what the heck this is happening...

daniel5151, 2021-10-11:
Indeed, it seems you're right!

I'm not sure why I didn't try this out sooner, but if I spin up the in-repo `armv4t` example and execute `stepi 10000` in GDB, hitting Ctrl-C triggers the same unexpected packet branch as you're seeing! 

Now that I have an easy local repro, I can really dig into what's going on, though unfortunately, this week is a bit hectic for me, so I'm not sure when exactly I'll get the chance to dig into this issue... 

In any case, I can confirm this is _definitely_ an issue on `gdbstub`'s end.

If you'd like to take a crack at fixing this issue, I'd be happy to review any code sent my way. Otherwise, give me some time, and I'll definitely fix this issue 😅

gz, 2021-10-11:
Nice! 
It seems like some clarification in the GDB documentation would be in order for this case... at least I couldn't find it there, the gdbserver implementation seems to be the documentation in this case :) 
FWIW deferring the Ctrl+C packet until we're in deferred stop state again seems like something that would be quite simple to support with the explicit state machine model. But it does make me wonder if this is the only packet that is a special case :)

gz, 2021-11-04:
Ok  --I have a test now that does Ctrl+C & continue and happy to report that it's working as expected on dev/0.6 :+1:

gz, 2021-11-04:
Got excited too early. I still have some issue with next (aka doing single-stepping) and sending Ctrl+C during that time. Will need to investigate further.

gz, 2021-11-04:
Ok nvm, I figured that was my bug -- when single-stepping I accidentially reported StepDone instead of SIGINT to `interrupt_handled`... Now it works with next/single-step and Ctrl+c too :+1:

closed 2021-11-04 by gz

daniel5151, 2021-11-04:
Fantastic! 

It sure was a long journey to get this fixed, but I'm glad we got it working in the end!

All that's left is the arduous task of documenting all of these APIs, updating the changelog, writing a 0.5 migration guide, and publishing to crates.io... At this rate, it's looking more and more likely that all that is going to have to happen sometime in December over the holiday season 😅 

Cheers!