daniel5151/gdbstub #111: Allow `read_addrs()` to send fewer bytes than requested

geigerzaehler, 2022-10-19:
GDBs `m` command allows the stub to respond with fewer bytes than requested if part of the memory is not accessible. Implementations of `SingleThreadBase::read_addrs()` cannot make use of that feature. If an implementation writes fewer bytes to the `data` argument the remaining bytes will be zero.

(Happy to provide a PR for this if you want me to.)

daniel5151, 2022-10-22:
Yep, this is certainly functionality `gdbstub` should support.

Unfortunately, this will require a breaking API change, as the current `read_addrs` API returns a `TargetResult<(), Self>`, whereas we'd want to return a `TargetResult<usize, Self>`, where the `usize` indicates the number of bytes written into the read buffer.

The PR should be pretty easy to implement, so if you're interested in taking a shot at getting this working, feel free to open one that targets the `dev/0.7` branch!

daniel5151, 2023-11-24:
Closed now that gdbstub 0.7 has been released

PR #115 (2022-12-07): `read_addrs()` can return fewer bytes than requested
### Description

The `read_addrs()` method for targets now returns a number of bytes that were read. This allows a target to gracefully signal the client that memory at an address range is not accessible.

I’ve implemented this functionality for the armv4t example to be able to test it.

Closes #111.

### API Stability

This PR requires a breaking API change.

### Checklist

- Implementation
  - [x] `cargo build` compiles without `errors` or `warnings`
  - [x] `cargo clippy` runs without `errors` or `warnings`
  - [x] `cargo fmt` was run
  - [... cut]
closed 2023-11-24 by daniel5151