daniel5151/gdbstub #123: Misleading comment in `state_machine`?

xobs, 2023-02-28:
In the documentation for `incoming_data()` in the `state::Running` state, the comment is annotated as such:

https://github.com/daniel5151/gdbstub/blob/d22eea8cc11db6fcd3ceb3bb1bb1f8cac88bde97/src/stub/state_machine.rs#L274-L302

However, contrary to the comment, it seems as though this performs state transitions as normal, and it doesn't always return a state that's `state::Running`.

Is this comment incorrect? Does it need to be updated?

daniel5151, 2023-02-28:
Yep, that's definitely wrong!
I believe I left that in from an earlier iteration of the code, and forgot to remove it.

Thanks for brining it to my attention. No need to send a PR or anything - I'll push up a commit fixing this shortly.

daniel5151, 2023-03-25:
closed via d69da3a

closed 2023-03-25 by daniel5151