daniel5151/gdbstub #140: Fails to compile for atmega328p (Arduino Uno)

konnorandrews, 2023-08-23:
I was experimenting with implementing a tool similar to https://github.com/jdolinay/avr_debug/tree/master but using this crate. However, I ran into the following error from LLVM:
```
LLVM ERROR: Cannot select: 0x2726a9bf9a8: i8 = ROL 0x2726a9bd558, C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:224:34 @[ C:\Users\konnor\.rustup\toolchains\nightly-2023-08-08-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1075:29 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:222:41 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:152:47 ] ] ]
  0x2726a9bd558: i8 = ROL 0x2726a9c8c38, C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:224:34 @[ C:\Users\konnor\.rustup\toolchains\nightly-2023-08-08-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1075:29 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:222:41 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:152:47 ] ] ]
    0x2726a9c8c38: i8 = ROL 0x2726a9c8ed8, C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:224:34 @[ C:\Users\konnor\.rustup\toolchains\nightly-2023-08-08-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1075:29 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:222:41 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:152:47 ] ] ]
      0x2726a9c8ed8: i8 = ROL 0x2726a9c5cf8, C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:224:34 @[ C:\Users\konnor\.rustup\toolchains\nightly-2023-08-08-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1075:29 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:222:41 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:152:47 ] ] ]
        0x2726a9c5cf8: i8 = add 0x2726a9c5ba8, Constant:i8<-67>, C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:224:34 @[ C:\Users\konnor\.rustup\toolchains\nightly-2023-08-08-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1075:29 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:222:41 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:152:47 ] ] ]
          0x2726a9c5ba8: i8,ch = load<(load (s8) from %ir.835, !alias.scope !20409, !noalias !15997)> 0x27266b771f8, 0x2726a9cc828, undef:i16, C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:224:34 @[ C:\Users\konnor\.rustup\toolchains\nightly-2023-08-08-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1075:29 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:222:41 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:152:47 ] ] ]
            0x2726a9cc828: i16,ch = CopyFromReg 0x27266b771f8, Register:i16 %247, C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:224:34 @[ C:\Users\konnor\.rustup\toolchains\nightly-2023-08-08-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\option.rs:1075:29 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:222:41 @[ C:\Users\konnor\.cargo\registry\src\index.crates.io-6f17d22bba15001f\gdbstub-0.6.6\src\stub\core_impl.rs:152:47 ] ] ]
              0x2726a9c44e8: i16 = Register %247
            0x2726a9cb478: i16 = undef
          0x2726a9bcd08: i8 = Constant<-67>
In function: _ZN7gdbstub4stub9core_impl24GdbStubImpl$LT$T$C$C$GT$13handle_packet17hba99f83e6a45ff72E
```

This is probably outside the scope of this project, but thought I would document it if anyone else tries to use it on a microcontroller.

daniel5151, 2023-08-23:
Indeed, this looks like a compiler bug of some sort...
IIRC the AVR target is still fairly new, so it's not _too_ surprising to have run into something like this.

Two follow ups:
1. If you're feeling generous, consider sending a minimal repro of this behavior upstream to the Rust project to see if someone can dig a little deeper into why LLVM is getting angry here.
2. You can try futzing around with the `gdbstub` code that's triggering this issue, since you _might_ be able to "massage" the code into a form that doesn't crash the compiler, which would unblock your particular use-case

I'll leave this open for a bit, though I suspect that I'll end up closing it after some period of inactivity, since unless you end up digging into (2) and finding some easy way to unblock this, it's unlikely there's much I'd be able to do from the `gdbstub` side of things to fix this.

Thanks for letting me know!

daniel5151, 2023-08-30:
Let me know if you ever decide to circle back here, otherwise, I'm closing this out for now.

closed 2023-08-30 by daniel5151