daniel5151/gdbstub #151: About response to 'qSupported'

rayc345, 2024-08-12:
Hello Daniel.
I compare the output of gdbstub and openocd and I found the difference partly lay in the response of qSupported.
 OpenOCD:
write: $qSupported:vContSupported+#6c
read: $PacketSize=4000;qXfer:memory-map:read+;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+#02

GDBStub:
write: $qSupported:vContSupported+#6c
read: $PacketSize=1000;vContSupported+;multiprocess+;QStartNoAckMode+;hwbreak+;qXfer:features:read+;qXfer:memory-map:read+#72

I read the source code of OpenOCD and found it never returned 'multiprocess+' instead it always return 'qXfer:threads:read+'.
I think 'multiprocess' only applies to target with OS like Linux, for embedded targets this should not exist. I want to know if we need to reevaluate the assumption in code:
"// these are the few features that gdbstub unconditionally supports
res.write_str(concat!(";vContSupported+", ";multiprocess+",))?;" // base.rs:115

I have some other problems when using probe-rs and I am digging where the problem lies in probe-rs or gdbstub.

daniel5151, 2024-08-12:
> I read the source code of OpenOCD and found it never returned 'multiprocess+' instead it always return 'qXfer:threads:read+'.

I'm not sure "instead" is the right term here. These are two totally independent features.

`qXfer:threads:read+` is a optional protocol feature that can be used to obtain thread info in XML form (as described via https://sourceware.org/gdb/current/onlinedocs/gdb.html/Thread-List-Format.html#Thread-List-Format).

No one has implemented this particular extension in `gdbstub`, so if this is something you're interested in supporting, feel free to send in a PR adding support for it :)

> I think 'multiprocess' only applies to target with OS like Linux, for embedded targets this should not exist. I want to know if we need to reevaluate the assumption in code:

That's not correct.

As per the docs (emphasis added): https://sourceware.org/gdb/current/onlinedocs/gdb.html/General-Query-Packets.html#multiprocess-extensions

> The remote stub understands the multiprocess extensions to the remote protocol syntax. The multiprocess extensions affect the syntax of thread IDs in both packets and replies (see [thread-id syntax](https://sourceware.org/gdb/current/onlinedocs/gdb.html/Packets.html#thread_002did-syntax)), and add process IDs to the ‘D’ packet and ‘W’ and ‘X’ replies. **Note that reporting this feature indicates support for the syntactic extensions only, not that the stub necessarily supports debugging of more than one process at a time.** The stub must not use multiprocess extensions in packet replies unless GDB has also indicated it supports them in its ‘qSupported’ request.

In other words - `gdbstub` will always report that it supports this extension, and the protocol-level syntax changes that it implies.

...that said, if the client does _not_ report support for this extension, `gdbstub` will properly perform "feature negotiation" to avoid sending over packets in the multiprocess format. See code at https://github.com/daniel5151/gdbstub/blob/
5b9b415/src/stub/core_impl/base.rs#L95

* * * 

Please let me know if you have any further questions :)

rayc345, 2024-08-16:
OK, thanks Daniel for your good explanation! 
I think currently we can keep the implementation.

closed 2024-08-16 by rayc345