daniel5151/gdbstub #153: Supports core::error::Error for Rust 1.81+ as a bound on gdbstub::target::Target::Error

ultimaweapon, 2024-10-05:
My application rely on `core::error::Error::source` to propagate the errors from the lower-level to the top-level. It would be great if `gdbstub` provide a support this!

daniel5151, 2024-10-06:
Ahh, that's right - Error is finally in core!
How exciting!

I would be happy to accept a PR that switches `gdbstub`'s error types over to using the `Error` type in `core` instead of `std`.

One caveat: I would prefer landing this behind a new off-by-default `core_error` feature, so as to ensure that all current users of `gdbstub`  0.7.x are able to update to new revisions of `gdbstub` without running into MSRV issues. Later, when it comes time to cut a `gdbstub` 0.8 release, I'll remove the feature, and require a MSRV of >1.81.

ultimaweapon, 2024-10-06:
Noted. I'll create a PR once ready.

PR #154 (2024-10-24): feature to use core::error::Error instead of std::error::Error
### Description

Not sure if the validation outputs are required because it is a small change.

Closes #153.

### API Stability

- [x] This PR does not require a breaking API change

### Checklist

- Documentation
  - [x] Ensured any public-facing `rustdoc` formatting looks good (via `cargo doc`)
  - [ ] (if appropriate) Added feature to "Debugging Features" in README.md
closed 2024-10-24 by daniel5151