daniel5151/gdbstub #166: File Paths collision on a case-insensitive filesystem

magravel, 2025-03-03:
Does not build on windows because two files has the same case insensitive name.

# Git Clone Error
[image]

# Build Error
[image]

daniel5151, 2025-03-03:
Uh oh. That's not good.
I recall running into this issue early on in `gdbstub`'s development, with some of the basic packets (e.g: `m` vs. `M`, etc...).

Will fix this ASAP, publish 0.7.5, and yank 0.7.4.

Thank you for the report!

daniel5151, 2025-03-03:
Resolved via cac18dca2d27cebb06513582e403f2964970f330. 0.7.5 is now published, with this fix.

closed 2025-03-03 by daniel5151