daniel5151/gdbstub #173: NoStd No Alloc build failures

AdamKMeller, 2025-08-08:
While working in a no std environment with alloc and I tried building the latest (0.7.6) with various feature flags and found some compilation errors. Namely, the alloc feature is seemingly required for the managed_vec.rs file to compile, erroring out on line 24 when alloc is not present. Adding alloc prevents this error, however, alloc with noStd seems to create some ambiguity in the tracepoints.rs (added in version 0.7.4) which will create a compiler error. I was only able to get a noStd build to work on version 0.7.3 with alloc. 

With version "=0.7.6", default-features = false but no alloc, an error would be thrown:
```
 error[E0004]: non-exhaustive patterns: `&mut &mut ManagedSlice::Owned(_)` not covered
  --> ... /.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/gdbstub-0.7.6/src/util/managed_vec.rs:24:19
   |
24 |             match &mut self.buf {
   |                   ^^^^^^^^^^^^^ pattern `&mut &mut ManagedSlice::Owned(_)` not covered
   |

```

By adding the alloc feature ( version = "=0.7.6", default-features = false, features = [ "alloc"]) the error would go away but another one would come up in the tracepoints added in version 0.7.4:
```
error[E0599]: no method named `to_owned` found for enum `ManagedSlice<'_, u8>` in the current scope
   --> .../ .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/gdbstub-0.7.6/src/target/ext/tracepoints.rs:124:51
    |
124 |             bytes: ManagedSlice::Owned(self.bytes.to_owned()),
    |                                                   ^^^^^^^^ method not found in `ManagedSlice<'_, u8>`
    |
    = help: items from traits can only be used if the trait is in scope
help: trait `ToOwned` which provides `to_owned` is implemented but not in scope; perhaps you want to import it
    |
18  + use crate::alloc::borrow::ToOwned;
    |
```

In order to get a properly compiling version of gdbstub with noStd I had to roll back to version 0.7.3 before tracepoints and include alloc (I also added core_error but this does not have an impact on the compilation as far as I can see).

bbbbbq, 2025-08-08:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

daniel5151, 2025-08-08:
Ah, looks like we might've missed a feature gate somewhere...
I can repro the issue locally by tweaking the `example_no_std` example in-tree, and I think I have a fix.

Can you add the following lines to `tracepoints.rs`, and let me know if it fixes things for you?

```rust
#[cfg(feature = "alloc")]
use crate::alloc::borrow::ToOwned;
```

If that works, feel free to send in a PR that fixes this, and I can cut 0.7.7 with the fix.

AdamKMeller, 2025-08-14:
Yes, adding the gated import fixed the issue for all flags on a fresh version of gdbstub. The repo builds as expected with it added. Added PR https://github.com/daniel5151/gdbstub/pull/174

Thanks Daniel!

PR #174 (2025-08-15): Fix to tracepoints no_std per daniel5151's rec in issue 173
### Description

Adds a gated import of toOwned which allows tracepoints to compile properly in a no_std environment. 

Closes #173  

### API Stability

- [x] This PR does not require a breaking API change

### Checklist

- Documentation
  - [ ] Ensured any public-facing `rustdoc` formatting looks good (via `cargo doc`)
  - [ ] (if appropriate) Added feature to "Debugging Features" in README.md
- Validation
  - [ ] Included output of running `examples/armv4t` with `RUST_LOG=trace` + any relevant GDB output under the "Validation" sect [... cut]
closed 2025-08-15 by daniel5151